RobotLocomotion/drake #1256: Consider adding a dummy viewer to the build servers

rdeits, 2015-08-13:
We've had errors in some examples creep in because the build servers disable their visualizers and therefore don't test anything in the visualization pipeline. We might want to create a way for the build servers to go through the motions of launching and drawing to the viewers, even if they don't actually display anything. 

One way to do this would be to install [Xvfb](https://en.wikipedia.org/wiki/Xvfb) on drake006, following the instructions on http://kovyrin.net/2007/10/01/how-to-run-gui-programs-on-a-server-without-any-monitor/
I believe that would let us launch drake_viewer so that it can listen to LCM draw commands. Instead of drawing to a monitor, it would just write to the virtual frame buffer provided by xvfb.

rdeits, 2015-08-13:
Also discussed here: http://askubuntu.com/questions/50599/how-do-you-run-a-gui-application-without-gui-gui-application-as-daemon-on-headl

patmarion, 2015-08-18:
Why not just put some resistors into the VGA port to that the X server detects a monitor?

http://ubuntuforums.org/showthread.php?t=1452600&page=2&p=9411761#post9411761

Kidding.  I've used Xvfb to run the director visualizer, that does work.  It's also possible to write an xorg.conf file to force the X server to start without detecting a monitor.

rdeits, 2015-08-18:
Hahaha. That's awesome.

RussTedrake, 2015-08-27:
fwiw - the situation gets a lot better after the fix in #1288.  when the visualizers are disabled, I have a "NullVisualizer" that gets constructed.  Before, that NullVisualizer took the state frame as input, while the BotVisualizer started taking only the position frame as input.  The bugs that crept in were the missing transforms from the more complicated atlas frames back to the position frame for display.  The NullVisualizer now takes the position frame as input and catches these bugs.

Running the actual visualizer would only help a little -- it would catch crashes in our actual visualizer classes.  I do agree that this would be good, but it's not at the top of my stack.  :)

RussTedrake, 2016-05-01:
fwiw -- this could now be tested by simply removing the g_disable_visualizers logic in the add_matlab_test script in drake/CMakeLists.txt .

patmarion, 2016-05-01:
If you need to run drake-visualizer on a build server without a display, it is possible to do so with xvfb.  To see how it's done on travis-ci to run director tests with visualization, see:

https://github.com/RobotLocomotion/director/blob/master/distro/travis/before_install.sh
https://github.com/RobotLocomotion/director/blob/master/.travis.yml

david-german-tri, 2016-07-18:
duplicate of #2767

closed 2016-07-18 by david-german-tri