RobotLocomotion/drake #1261: disable botvis by default, and build director instead

RussTedrake, 2015-08-16:
as described here: https://github.com/RobotLocomotion/director/issues/41

patmarion, 2015-08-17:
So far, I've added a libbot option to allow building the lcm-gl library without the bot-vis dependency.  There isn't an option to disable bot-vis completely, a lot of bot libs still require bot-vis.  But it is possible to run make just in the lcm-gl subdirectory and skip over everything else.  One option would be to have drake's CMakeLists.txt handle the custom build configuration of libbot.  Another option would be to let the director handle it, could even compile lcm-gl statically and don't even bother installing it, so it's internal to the director build.

RussTedrake, 2015-08-17:
I'm ok with either of those.

If it's handled by drake, then there should still be an option to use libbot the old way. but i also like the idea of having director grab the lcmgl as needed by itself.

RussTedrake, 2015-09-29:
i think people are going to have a bad experience building on mac with the default options until we resolve this.

tarquasso, 2015-09-30:
for now, what non-default option shall I choose?

closed 2015-12-16 by RussTedrake