RobotLocomotion/drake #1263: Travis builds fail with missing dart config (can't see why it only fails sometimes)

RussTedrake, 2015-08-16:
Using worker: worker-linux-c55820f2-1.bb.travis-ci.org:travis-linux-1
system_info
Build system information
Build language: cpp
Build image provisioning date and time
Wed Feb  4 18:22:50 UTC 2015
Operating System Details
Distributor ID:	Ubuntu
Description:	Ubuntu 12.04 LTS
Release:	12.04
Codename:	precise
Linux Version
2.6.32-042stab090.5
Cookbooks Version
23bb455 https://github.com/travis-ci/travis-cookbooks/tree/23bb455
GCC version
gcc (Ubuntu/Linaro 4.6.3-1ubuntu5) 4.6.3
Copyright (C) 2011 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
LLVM version
clang version 3.4 (tags/RELEASE_34/final)
Target: x86_64-unknown-linux-gnu
Thread model: posix
Pre-installed Ruby versions
ruby-1.9.3-p551
Pre-installed Node.js versions
v0.10.36
Pre-installed Go versions
1.4.1
Redis version
redis-server 2.8.19
riak version
2.0.2
MongoDB version
MongoDB 2.4.12
CouchDB version
couchdb 1.6.1
Neo4j version
1.9.4
Cassandra version
2.0.9
RabbitMQ Version
3.4.3
ElasticSearch version
1.4.0
Installed Sphinx versions
2.0.10
2.1.9
2.2.6
Default Sphinx version
2.2.6
Installed Firefox version
firefox 31.0esr
PhantomJS version
1.9.8
ant -version
Apache Ant(TM) version 1.8.2 compiled on December 3 2011
mvn -version
Apache Maven 3.2.5 (12a6b3acb947671f09b81f49094c53f426d8cea1; 2014-12-14T17:29:23+00:00)
Maven home: /usr/local/maven
Java version: 1.7.0_76, vendor: Oracle Corporation
Java home: /usr/lib/jvm/java-7-oracle/jre
Default locale: en, platform encoding: UTF-8
OS name: "linux", version: "2.6.32-042stab090.5", arch: "amd64", family: "unix"
git.checkout
3.56s$ git clone --depth=50 git://github.com/RobotLocomotion/drake.git RobotLocomotion/drake
Cloning into 'RobotLocomotion/drake'...
remote: Counting objects: 6142, done.
remote: Compressing objects: 100% (3880/3880), done.
[... 1288 lines cut ...]
0.01s$ cd drake/pod-build
/home/travis/build.sh: line 41: cd: drake/pod-build: No such file or directory
The command "cd drake/pod-build" exited with 1.
1.66s$ ctest -j 1 -D Experimental -C $BUILD_TYPE --output-on-failure --timeout 1000 --track Continuous
Cannot find file: /home/travis/build/RobotLocomotion/drake/DartConfiguration.tcl
   Site: 
   Build name: 
Create new tag: 20150816-1946 - Continuous
Configure project
Cannot find ConfigureCommand key in the DartConfiguration.tcl
Cannot find file: /home/travis/build/RobotLocomotion/drake/DartConfiguration.tcl
Build project
Cannot find MakeCommand key in the DartConfiguration.tcl
Cannot find file: /home/travis/build/RobotLocomotion/drake/DartConfiguration.tcl
Test project /home/travis/build/RobotLocomotion/drake
No tests were found!!!
Cannot find file: /home/travis/build/RobotLocomotion/drake/DartConfiguration.tcl
Performing coverage
Cannot find source Python file corresponding to: /home/travis/.gimme/versions/go1.4.1.linux.amd64/blog/content/cover/pkg.cover
 Cannot find any coverage files. Ignoring Coverage request.
Cannot find file: /home/travis/build/RobotLocomotion/drake/DartConfiguration.tcl
Cannot find BuildDirectory  key in the DartConfiguration.tcl
Errors while running CTest
The command "ctest -j 1 -D Experimental -C $BUILD_TYPE --output-on-failure --timeout 1000 --track Continuous" exited with 70.
Done. Your build exited with 1.

rdeits, 2015-08-16:
The actual error is earlier, during the build, and is caused by the issue that was resolved in https://github.com/RobotLocomotion/lcm-pod/pull/8 

It looks like some branches don't yet have the most up-to-date version of the LCM pod yet.

rdeits, 2015-08-16:
Specifically, the error is that the LCM pod's build_python target calls "time", which isn't installed on travis. That pull request just removes the call to "time".

RussTedrake, 2015-08-16:
> The actual error is earlier, during the build, and is caused by the issue that was resolved in RobotLocomotion/lcm-pod#8

Ah, good.  That was just merged.  (Just after this build started).  So future builds should be aok

rdeits, 2015-08-16:
Excellent!

closed 2015-08-16 by RussTedrake