RobotLocomotion/drake #1272: A class for motor dynamics in drake

hosseinZabihi, 2015-08-20:
For implementation of Nao robot in drake, I faced with a challenge to implement the joints motor. I could not find a straight way to do that!
I know that by using urdf we can define friction and damping for each joint, but I want to have a seperate system as a motor which can get the states and other aspects of it.
Firstly, I looked at RigidBodyActuator but it seems incomplete and the only way comes to my mind is to build a linear system contains actuator joints model and cascade it to our robot model. A motor is a common element in every robot and I guess there must be better ways to include it in a robot model.
Is there any class or tool in drake that eases implementing motors, or is there any example guides me to implement this class efficiently?

tkoolen, 2015-08-21:
> the only way comes to my mind is to build a linear system contains actuator joints model and cascade it to our robot model
Yes, that's probably the way to go. RigidBodyActuator would need a big overhaul to support your use case. We should probably do that at some point, but for now a cascade is the only option.

tkoolen, 2015-08-21:
Related: https://github.com/RobotLocomotion/drake/issues/551

RussTedrake, 2015-08-27:
agreed.  i have done it myself by cascading the actuator model.  as @tkoolen points out, we'd actually like to move even more in that direction.  But it would definitely be great to have classes for the standard motor models that could even get pulled from urdf.  

Will close this for now.  But looking forward to hearing how it works for you.

closed 2015-08-27 by RussTedrake