RobotLocomotion/drake #1276: mac java strikes again?

RussTedrake, 2015-08-22:
I'm pretty sure I just wasted 4 hours thinking I had introduced a memory bug in my c++ code... now I'm *fairly certain* that thing that changed was that I upgraded CLion to the newest version (with Java bundled).  Matlab 2015a now reliably crashes when I run 
```
matlab -r "addpath_drake; cd examples/Pendulum; v = PendulumVisualizer;"
```
This happens on master, and on builds going back a few weeks.  Matlab 2014a is ok, but does announce this at startup:
```
Exception "java.lang.ClassNotFoundException: com.intellij.codeInsight.editorActions.FoldingData"while constructing DataFlavor for: application/x-java-jvm-local-objectref; class=com.intellij.codeInsight.editorActions.FoldingData
Exception "java.lang.ClassNotFoundException: com.intellij.codeInsight.editorActions.FoldingData"while constructing DataFlavor for: application/x-java-jvm-local-objectref; class=com.intellij.codeInsight.editorActions.FoldingData
```

posting here in case anybody can confirm or if you see the same thing.

tkoolen, 2015-08-23:
Yeah I get that and I'm not too worried about it. It is related to CLion. I'm not sure why it happens.

RussTedrake, 2015-08-23:
yikes.  when did you start getting it?  (did you just notice now that you look)

how do you know it’s related to CLion?  is there a good workaround?

tkoolen, 2015-08-23:
Oh sorry, I didn't fully process the part about the Matlab crash. I don't know if that's related to CLion and I hadn't noticed that before. I can reproduce it in Matlab 2015a though. I get this message on the ocmmand line:
```
objc[17120]: Class JavaLaunchHelper is implemented in both /Applications/MATLAB_R2015a.app/sys/java/jre/maci64/jre/bin/java and /Applications/MATLAB_R2015a.app/sys/java/jre/maci64/jre/lib/jli/libjli.dylib. One of the two will be used. 
Which one is undefined.
```
That seems unrelated to CLion.

I know the exceptions on startup are related to CLion because of the class name: `com.intellij.*`. Intellij is the former name of JetBrains, the company that develops CLion. I only get it when I debug Matlab through CLion by the way, not when I just start it from the command line.

RussTedrake, 2015-08-23:
to confirm, you see the crash in master (not my dev branch)… correct?

tkoolen, 2015-08-23:
Yeah, in master.
What's interesting is that if you place a breakpoint at line 10 of PendulumVisualizer (the end of the constructor), create a PendulumVisualizer and then *step in* twice, Matlab won't crash. If you step or continue, Matlab will crash.

tkoolen, 2015-08-23:
This is probably related:
https://stackoverflow.com/questions/20794751/class-javalaunchhelper-is-implemented-in-both-one-of-the-two-will-be-used-whic

The first answer states:
> From what I've found online, this is a bug introduced in JDK 1.7.0_45. It appears to also be present in JDK 1.7.0_60. A bug report on Oracle's website states that, while there was a fix, it was removed before the JDK was released.

It just so happens to be that
```
/Applications/MATLAB_R2015a.app/sys/java/jre/maci64/jre/bin/java -version
```
results in
```
java version "1.7.0_60"
Java(TM) SE Runtime Environment (build 1.7.0_60-b19)
Java HotSpot(TM) 64-Bit Server VM (build 24.60-b09, mixed mode)
```

So I think it's a bug in the version of Java that Matlab 2015a bundles.

RussTedrake, 2015-08-23:
fantastic.  thanks for looking at it.  can’t believe how long I wasted yesterday thinking it was a c++ memory issue.  :)

tkoolen, 2015-08-23:
I tried unsuccesfully to get around it by specifying the MATLAB_JAVA environment variable to change the VM as in
https://www.mathworks.com/matlabcentral/answers/103056-how-do-i-change-the-java-virtual-machine-jvm-that-matlab-is-using-for-mac-os

If I use
```
export MATLAB_JAVA=/Library/Java/JavaVirtualMachines/jdk1.8.0_40.jdk/Contents/Home/jre
```
I still get the crash, but now the message on the command line is
```
objc[22327]: Class JavaLaunchHelper is implemented in both
/Library/Java/JavaVirtualMachines/jdk1.8.0_40.jdk/Contents/Home/jre/bin/java and
/Applications/MATLAB_R2015a.app/sys/java/jre/maci64/jre/lib/jli/libjli.dylib. One of the two will be used. 
Which one is undefined.
```
i.e. it is using the specified version of Java, but same issue.

If I use
```
export MATLAB_JAVA=/Library/Java/JavaVirtualMachines/jdk1.7.0_17.jdk/Contents/Home/jre
```
i.e. a version before the bug was introduced, I no longer get a printout to the console, but Matlab still crashes...

tkoolen, 2015-08-23:
I get the same crash (with no message printed to console) in Ubuntu 14.04 with Matlab R2015a. Could be an R2015a specific issue then.

tkoolen, 2015-08-23:
Also interesting: this does not crash Matlab:
```
matlab -r "addpath_drake; cd examples/Pendulum; bla = LCMCoordinateFrame('PendulumState','drake.examples.Pendulum.lcmt_pendulum_x','x',{'theta','thetadot'}); v = PendulumVisualizer;"
```

DiitsGp, 2015-10-23:
Found this as the first result when google searching! I installed Clion yesterday and it forced me to install Mac Java. Now I get this after running addpath_pods and addpath_drake:

Adding /Users/diits/newdrake/drake-distro/build/matlab to the matlab path
Exception "java.lang.ClassNotFoundException: com/intellij/codeInsight/editorActions/FoldingData"while constructing DataFlavor for: application/x-java-jvm-local-objectref; class=com.intellij.codeInsight.editorActions.FoldingData
Exception "java.lang.ClassNotFoundException: com/intellij/codeInsight/editorActions/FoldingData"while constructing DataFlavor for: application/x-java-jvm-local-objectref; class=com.intellij.codeInsight.editorActions.FoldingData
>>

tkoolen, 2015-10-23:
CLion should come packaged with a JDK on OSX, so I don't think you should have to install Mac Java.

But I get those warnings too and they don't cause any trouble. The warnings were kind of a red herring for this issue.

tkoolen, 2015-10-23:
I can't reproduce this issue on 2015b (anymore?) by the way.

geraneum, 2015-11-21:
I wandered here following the same exception with Matlab 2015a. When PyCharm is open, I get this exception. Also happens with IntelliJ.
There is also a [SO question](http://stackoverflow.com/questions/7846657/why-how-is-intellij-causing-debug-output-in-netbeans) about this.

tkoolen, 2015-11-22:
This issue is really about a Matlab crash, which I believe is not related to the exceptions on startup. Again, I think the exceptions don't cause any real trouble.

RussTedrake, 2016-01-23:
although it's unsatisfying, i haven't seen this in a while (and don't have any more leads).  will close for now

closed 2016-01-23 by RussTedrake

avalenzu, 2016-01-24:
If you do encounter MATLAB crashes that are unrelated to drake C++ code in
the future, please don't hesitate to submit them to TMW.