RobotLocomotion/drake #1281: installation

ghuber-github, 2015-08-24:
hi guys,

just wanted to let you know that for my installation i encountered problems with ccmake (on the 'make options' command) and glut.h (on a 'make' command). So i was missing
```
$ sudo apt-get install cmake-curses-gui freeglut3-dev
```
to make it work.

Best regards from Munich,
Gerold

RussTedrake, 2015-08-24:
I will add them.  thank you!

RussTedrake, 2015-08-24:
do you happen to know which external needed the freeglut3-dev?

ghuber-github, 2015-08-25:
Hi Russ,

1) I just removed the freeglut3-dev package and rebuilt everything again 
from scratch to check and it seems that the bot2-vis library from the 
external libbot projects needs it:
--------------------------------------------------------
Scanning dependencies of target bot2-vis
[  3%] Building C object src/bot_vis/CMakeFiles/bot2-vis.dir/viewer.c.o
/home/ghuber/drake/drake-distro/externals/libbot/bot2-vis/src/bot_vis/viewer.c:18:21: 
fatal error: GL/glut.h: No such file or directory
  #include <GL/glut.h>
--------------------------------------------------------

2) Btw, the whole building process ends up in the Matlab console prompt, 
because on my system (xubuntu 14.04) the Matlab option "wait" doesn't 
seem to exist. Moreover the function 'spot_pods_install' is not found. 
(I don't now exactly if i need that one, just thought you might be 
interested):
--------------------------------------------------------
Unrecognized MATLAB option "wait".

                                                   < M A T L A B (R) >
                                         Copyright 1984-2015 The 
MathWorks, Inc.
                                         R2015a (8.5.0.197613) 64-bit 
(glnxa64)
                                                    February 12, 2015

To get started, type one of these: helpwin, helpdesk, or demo.
For product information, visit www.mathworks.com.

     Academic License

Undefined function or variable 'spot_pods_install'.

 >>
--------------------------------------------------------

After quitting the nodesktop version of Matlab the installation 
completes fine!

3) while testing (make test) the first test failed (on command r = 
RigidBodyManipulator('Acrobot.urdf');)

--------------------------------------------------------
Error using xmlstringinput (line 42)
File Acrobot.urdf not found

Error in xmlread (line 76)
    fileName = xmlstringinput(fileName,true);

Error in RigidBodyManipulator/addRobotFromURDF (line 43)
urdf = xmlread(urdf_filename);

Error in RigidBodyManipulator (line 66)
           obj = 
addRobotFromURDF(obj,filename,zeros(3,1),zeros(3,1),options);
--------------------------------------------------------

4) during the testing procedure (make test) i saw that it could not find 
gurobi and snopt_c. After activating them in 'make options' and running 
make again, I noticed that it seems the snopt.git is not online anymore:

--------------------------------------------------------
Cloning into 'snopt'...
Username for 'https://github.com': ghuber-github
Password for 'https://ghuber-github@github.com':
remote: Repository not found.
fatal: repository 'https://github.com/RobotLocomotion/snopt.git/' not found
--------------------------------------------------------

Best regards,
Gerold

-- 
Dipl.-Ing. Gerold Huber
Research Associate

Chair of Automatic Control Engineering (LSR)
TUM Department of Electrical, Electronic and Computer Engineering (EI)
Technische Universität München (TUM)
80333 Munich, Germany

Room No: N3509
Phone:   +49-89-289-23445
Fax:     +49-89-289-28340
E-Mail:  gerold.huber@tum.de
Web:     http://www.lsr.ei.tum.de

RussTedrake, 2015-08-25:
> On Aug 25, 2015, at 4:54 AM, ghuber-github <notifications@github.com> wrote:
> 
> Hi Russ,
> 
> 1) I just removed the freeglut3-dev package and rebuilt everything again 
> from scratch to check and it seems that the bot2-vis library from the 
> external libbot projects needs it:
> --------------------------------------------------------
> Scanning dependencies of target bot2-vis
> [ 3%] Building C object src/bot_vis/CMakeFiles/bot2-vis.dir/viewer.c.o
> /home/ghuber/drake/drake-distro/externals/libbot/bot2-vis/src/bot_vis/viewer.c:18:21: 
> fatal error: GL/glut.h: No such file or directory
> #include <GL/glut.h>

thank you.  it is actually already listed in libbot’s install_prereqs.sh
https://github.com/RobotLocomotion/libbot/blob/master/install_prereqs.sh <https://github.com/RobotLocomotion/libbot/blob/master/install_prereqs.sh>
so hopefully it would come in if you’ve run that (recursive) script?

perhaps we’ve introduced an error in that script with the new build system?  or perhaps an error in the instructions.  I can see one problem now:  we tell people to run install_prereqs before telling them to run make options, so if one adds an external with a new prereq, then would need to run install_prereqs again.  i’ll fix that next time I have a minute.

> --------------------------------------------------------
>
> 2) Btw, the whole building process ends up in the Matlab console prompt,
> because on my system (xubuntu 14.04) the Matlab option "wait" [...]

the -wait is a warning not an error.  i decided long ago it was acceptable instead of forking yet again for every OS / matlab version.  :)

the spot_pods_install is a problem.  You might not need spot, but it would be good to understand why that method is missing.  Did the git clone fail in that directory?

> 3) while testing (make test) the first test failed (on command r =
> RigidBodyManipulator('Acrobot.urdf');)
>
> --------------------------------------------------------
> Error using xmlstringinput (line [...]

that seems most likely to be a WORKING_DIRECTORY issue.  will have to figure out how to reproduce it to fix it.

> 4) during the testing procedure (make test) i saw that it could not find
> gurobi and snopt_c. After activating them in 'make options' and running
> make again, I noticed that it seems the snopt.git [...]

unfortunately, snopt requires a license.  the repo is there, but password protected.  we are allowed to redistribute code we’ve written which links statically with snopt (see #712 )

thanks for reporting these and helping to sort them out.

ghuber-github, 2015-08-26:
Hi Russ,

it took me all day, but I actually found the solutions

>
> > On Aug 25, 2015, at 4:54 AM, ghuber-github 
> >
> > Hi Russ,
> >
> > 1) I just removed the freeglut3-dev package and rebuilt everything 
> again
> > from scratch to check and it seems that the bot2-vis library from the
> > external libbot projects needs it:
> > --------------------------------------------------------
> > Scanning dependencies of target bot2-vis
> > [ 3%] Building C object src/bot_vis/CMakeFiles/bot2-vis.dir/viewer.c.o
> > 
> /home/ghuber/drake/drake-distro/externals/libbot/bot2-vis/src/bot_vis/viewer.c:18:21: 
>
> > fatal error: GL/glut.h: No such file or directory
> > #include <GL/glut.h>
>
>
> thank you. it is actually already listed in libbot’s install_prereqs.sh
> https://github.com/RobotLocomotion/libbot/blob/master/install_prereqs.sh 
> so hopefully it would come in if you’ve run that (recursive) script?
>
> perhaps we’ve introduced an error in that script with the new build 
> system? or perhaps an error in the instructions. I can see one problem 
> now: we tell people to run install_prereqs before telling them to run 
> make options, so if one adds an external with a new prereq, then would 
> need to run install_prereqs again. i’ll fix that next time I have a 
> minute.

You are right! The prereqs file of libbot lists freeglut3. However, 
After configuring the installation with 'make options' only the folders 
of the externals are created:
-------------------
ghuber@geronimo:~/drake/drake-distro2$ tree externals/
externals/
├── avl
├── bertini
├── bullet
├── director
│   └── distro
│       └── pods
│           └── drake-distro
├── eigen
├── gloptipoly
├── gtk
├── gurobi
├── install_prereqs.sh
├── iris
├── lcm
├── libbot
├── meshconverters
├── mosek
├── octomap
├── sedumi
├── signalscope
├── snopt
├── spotless
├── xfoil
└── yalmip
-------------------
so calling the install_prereqs.sh script after 'make options' doesn't 
help. The files (including the prereqs that installs freeglut3-dev) are 
downloaded during the make build. After the make error, the 
install_prereqs script can be called again, which then actually installs 
freeglut3 and freeglut3-dev.

> prompt,
>
> the -wait is a warning not an error. i decided long ago it was
> acceptable instead of forking yet again for every OS / matlab version. :)
>
> the spot_pods_install is a problem. You might not [...]
The git clone didn't have any problems. But I found the problem:
In the CMake generated file 
drake-distro/externals/spotless/pod-build/CMakeFiles/run.dir/build.make 
line 58, the command
 >>> cd /home/ghuber/drake/drake-distro2/externals/spotless && 
/usr/bin/matlab -wait -nosplash -nodesktop     -nodisplay -r 
"spot_pods_install('/home/ghuber/drake/drake-distro2/build/matlab','../../externals/spotless' 
);exit"

fails, because my matlab working directory on startup is always the 
directory from the last matlab run. So the beforehand called 'cd' 
command doesn't affect the matlab working directory. A quick fix that 
works in these cases could be to use 'cd' from within matlab:

 >>> /usr/bin/matlab -wait -nosplash -nodesktop -nodisplay -r "cd 
/home/ghuber/drake/drake-distro2/externals/spotless; 
spot_pods_install('/home/ghuber/drake/drake-distro2/build/matlab','../../externals/spotless' 
);exit"

This worked for me, and I think it maybe covers a more general case. Or 
maybe just a hint on the install instructions to not set this matlab 
behaviour in the instructions?

> that seems most likely to be a WORKING_DIRECTORY issue. will have to
> figure out how to reproduce it to fix it.
same problem here. (My) matlab loaded the last working directory and 
didn't care about the WORKING_DIRECTORY from the drake/CMakeList line 212:
-------------------------
208 if (MATLAB_FOUND)
209   add_test(NAME "RigidBodyManipulatorMemoryTest"
210         WORKING_DIRECTORY "${PROJECT_SOURCE_DIR}/examples/Acrobot"
211         COMMAND "matlab" #"${PROJECT_SOURCE_DIR}/cmake/matlab_clean.pl"
212         "-nosplash" "-nodisplay" "-wait" "-r" 
"addpath('${CMAKE_INSTALL_PREFIX}/matlab'); addpath_${POD_    NAME}; 
try, r = RigidBodyManipulator('Acrobot.urdf'); megaclear; catch ex, 
disp(getReport(ex,'extended')    ); disp(''); force_close_system; 
exit(1); end; force_close_system; exit(0)")
-----------------------

Maybe same fix here? Anyways, after preparing the working directory 
correctly, i dealt with another problem that "r = 
RigidBodyManipulator('Acrobot.urdf');" threw the error
drake/systems/plants/constructModelmex.mex64'

because matlabR2015 comes with libstdc++.so.6.0.17

After struggling for half a day how to convince matlab to use my newer 
local version of libstdc++6, I found the solution here:
https://www.mathworks.com/matlabcentral/newsreader/view_thread/255846
"One workaround to the problem that Matlab internally changes the 
LD_LIBRARY_PATH to prefer <MatlabPATH>/sys/os/<ARCH>/ is to set 
LD_PRELOAD when calling Matlab."
That works without messing up the matlab installation itself, which is 
handy if it runs on a server.

> find
> not found
>
>
> unfortunately, snopt requires a license. the repo is there, but
> password protected. we are allowed to redistribute code we’ve written
> which links statically with snopt (see #712 [...]

A last hint I found:
I tried a new clean reinstall to test a installation from scratch and it 
seems it is a bad idea to call the git clone folder 'drake' instead of 
'drake-distro'. Because if it is called drake, the first 'make' build is 
fine, but afterwards everything always throws up weird errors. Even a 
second 'make' call. If i name it 'drake-distro' the problem is gone.

Thanks for your help!

RussTedrake, 2015-08-27:
Fantastic.  Thanks for these.

### Re:  install_prereqs logic
@rdeits was in the process of updating that.  I will assign that one to him to make sure the new solution resolves this case.

### Re: your startup.m called cd
Good catch -- never considered that possibiity.  :)  I've inserted the manual cd call to the front of the spotless installer and the RigidBodyManipulatorMemoryTest.  I also made the change in the unit test scripts.  I spent a few minutes looking for other places, but didn't see any right now.

 ### libstdc++ version conflict
I've tried hard to make it so this wouldn't happen.  The build parses the mex linker flags, in the hopes that we link against the matlab versions of the library where appropriate.  But it's certainly possible that libstdc++ is getting pulled from another path that is earlier in the list of link directories.  If nothing else, having your fix documented here is useful.  What platform are you running on, precisely?

ghuber-github, 2015-08-27:
i hope i got it all covered:

```
ghuber@geronimo:~/git/drake-distro/drake$ cat /etc/*-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=14.04
DISTRIB_CODENAME=trusty
DISTRIB_DESCRIPTION="Ubuntu 14.04.3 LTS"
NAME="Ubuntu"
VERSION="14.04.3 LTS, Trusty Tahr"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 14.04.3 LTS"
VERSION_ID="14.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"

ghuber@geronimo:~/git/drake-distro/drake$ cat /proc/version
Linux version 3.13.0-49-generic (buildd@akateko) (gcc version 4.8.2 (Ubuntu 4.8.2-19ubuntu1) ) #83-Ubuntu SMP Fri Apr 10 20:11:33 UTC 2015

ghuber@geronimo:~/git/drake-distro/drake$ lscpu
Architecture:          x86_64
CPU op-mode(s):        32-bit, 64-bit
Byte Order:            Little Endian
CPU(s):                4
On-line CPU(s) list:   0-3
Thread(s) per core:    1
Core(s) per socket:    4
Socket(s):             1
NUMA node(s):          1
Vendor ID:             GenuineIntel
CPU family:            6
Model:                 58
Stepping:              9
CPU MHz:               3201.000
BogoMIPS:              6400.03
Virtualization:        VT-x
L1d cache:             32K
L1i cache:             32K
L2 cache:              256K
L3 cache:              6144K
NUMA node0 CPU(s):     0-3
```

rdeits, 2015-08-27:
Yup, the new installation procedure should fix this. Now a full installation means:

1. `make options` to choose which externals you want
1. `make download-all` which downloads all the externals and registers all of their `install_prereqs` scripts
1. `./install_prereqs.sh` which installs the prereqs for all the externals you chose
1. `make` to build and install everything locally

RussTedrake, 2015-08-28:
Thanks @rdeits -- I've updated the instructions on the wiki.

That still leaves the libstdc++ issue, and the concern about cloning into the directory drake (not sure i understand why that's a problem, but will test).

vtjeng, 2015-10-02:
I attempted to install the most recent public version of Drake on ubuntu, and ran `make test` after installation was complete. A large number of tests seemed to fail because of invalid MEX-files. 

A solution that worked for me was to modify the symlinks in my MATLAB installation, by first making sure that g++ 4.4 was installed:

    sudo apt-get install g++-4.4

And then modifying the symlinks in `/usr/local/MATLAB/R2014a/sys/os/glnxa64` to point to versions in `/usr/lib`, and not the same directory.

    sudo rm libgfortran.so.3
    sudo ln -s /usr/lib/x86_64-linux-gnu/libgfortran.so.3.0.0 libgfortran.so.3
    sudo rm libstdc++.so.6
    sudo ln -s /usr/lib/gcc/x86_64-linux-gnu/4.4/libstdc++.so libstdc++.so.6

Here is the information on my system

    vtjeng@vtjeng-Precision-M4700:~/drake-distro$ cat /etc/*-release
    DISTRIB_ID=Ubuntu
    vtjeng@vtjeng-Precision-M4700:~/drake-distro$ cat /etc/*-release
    DISTRIB_ID=Ubuntu
    DISTRIB_RELEASE=14.04
    DISTRIB_CODENAME=trusty
    DISTRIB_DESCRIPTION="Ubuntu 14.04.3 LTS"
    NAME="Ubuntu"
    VERSION="14.04.3 LTS, Trusty Tahr"
    ID=ubuntu
    ID_LIKE=debian
    PRETTY_NAME="Ubuntu 14.04.3 LTS"
    VERSION_ID="14.04"
    HOME_URL="http://www.ubuntu.com/"
    SUPPORT_URL="http://help.ubuntu.com/"
    BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"

    vtjeng@vtjeng-Precision-M4700:~/drake-distro$ cat /proc/version
    Linux version 3.19.0-25-generic (buildd@lgw01-20) (gcc version 4.8.2 (Ubuntu 4.8.2-19ubuntu1) ) #26~14.04.1-Ubuntu SMP Fri Jul 24 21:16:20 UTC 2015

    vtjeng@vtjeng-Precision-M4700:~/drake-distro$ lscpu
    Architecture:          x86_64
    CPU op-mode(s):        32-bit, 64-bit
    Byte Order:            Little Endian
    CPU(s):                8
    On-line CPU(s) list:   0-7
    Thread(s) per core:    2
    Core(s) per socket:    4
    Socket(s):             1
    NUMA node(s):          1
    Vendor ID:             GenuineIntel
    CPU family:            6
    Model:                 58
    Stepping:              9
    CPU MHz:               3696.144
    BogoMIPS:              6179.75
    Virtualization:        VT-x
    L1d cache:             32K
    L1i cache:             32K
    L2 cache:              256K
    L3 cache:              8192K
    NUMA node0 CPU(s):     0-7
    DISTRIB_RELEASE=14.04
    DISTRIB_CODENAME=trusty
    DISTRIB_DESCRIPTION="Ubuntu 14.04.3 LTS"
    NAME="Ubuntu"
    VERSION="14.04.3 LTS, Trusty Tahr"
    ID=ubuntu
    ID_LIKE=debian
    PRETTY_NAME="Ubuntu 14.04.3 LTS"
    VERSION_ID="14.04"
    HOME_URL="http://www.ubuntu.com/"
    SUPPORT_URL="http://help.ubuntu.com/"
    BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"

tkoolen, 2015-10-03:
This came from the DRC install instructions of course. We should probably add it as a FAQ item at least, possibly even in the standard installation instructions for Ubuntu.

RussTedrake, 2015-10-03:
It didn't use to be be necessary for drake.  What's changed?

I think it was the lcm c++ that required it...

vtjeng, 2015-10-05:
As a follow up, these were my installation options in `make options`. 

Modifying the symlinks reduced the number of errors in `make test` from approximately 200 to 40, with no errors that seem to be due to `MEX` files.

     AUTO_UPDATE_EXTERNALS            ON                                                                 
     CMAKE_BUILD_TYPE                 Release                                                            
     CMAKE_INSTALL_PREFIX             /usr/local                                                         
     EXAMPLES_LITTLEDOG               OFF                                                                
     MAKE_EXECUTABLE                  /usr/bin/make                                                      
     REMOVE_UNUSED_EXTERNALS          OFF                                                                
     WITH_ALL_PUBLIC_EXTERNALS        ON                                                                 
     WITH_ALL_SUPPORTED_EXTERNALS     OFF                                                                
     WITH_AVL                         OFF                                                                
     WITH_BERTINI                     OFF                                                                
     WITH_BULLET                      ON                                                                 
     WITH_DIRECTOR                    ON                                                                 
     WITH_EIGEN                       ON                                                                 
     WITH_GLOPTIPOLY                  OFF                                                                
     WITH_GUROBI                      OFF                                                                
     WITH_IRIS                        OFF                                                                
     WITH_LCM                         ON                                                                 
     WITH_LIBBOT                      ON                                                                 
     WITH_MESHCONVERTERS              OFF                                                                
     WITH_MOSEK                       OFF                                                                
     WITH_OCTOMAP                     OFF                                                                
     WITH_SEDUMI                      OFF                                                                
     WITH_SIGNALSCOPE                 OFF                                                                
     WITH_SNOPT                       OFF                                                                
     WITH_SNOPT_PRECOMPILED           OFF                                                                
     WITH_SPOTLESS                    ON                                                                 
     WITH_XFOIL                       OFF                                                                
     WITH_YALMIP                      OFF                                                                
     _build_dir                       /home/vtjeng/drake-distro/build                                    
     matlab                           /usr/bin/matlab

tkoolen, 2015-10-05:
(At least a part of) the rest are likely to be due to not selecting all externals in make options. I'd say that's about expected.

RussTedrake, 2015-12-16:
Recommendation:  we could put the list of all prereqs (given the selected externals) in the drake superbuild file.  We might also be able to get rid of the install_prereqs required to run download_all.  @patmarion feels strongly that installations should be a copy-and-paste so the user can send it to the command line.

rdeits, 2015-12-16:
As an additional piece of info: our Travis builds will start substantially faster (and be easier on their end) if we explicitly list our prerequisites instead of recursively discovering them. Travis can use their new architecture, in which each build lives in an isolated container, but not a full VM, to speed up builds, but that requires that we don't run any commands as root. Instead, we just list the apt packages we need in our travis.yml file. 

 We currently only use sudo to run `./install_prereqs.sh`, so eliminating that would be a minor benefit.

RussTedrake, 2015-12-17:
I'm ok with moving forward on this (and removing the install_prereqs throughout).

jamiesnape, 2016-03-23:
This will be fixed by #1862.

PR #1862 (2016-03-24): Ensure Clang/GCC are at least the minimum version
This closes #960, closes #1281, and closes #1649.

PTAL @RussTedrake.
closed 2016-03-24 by RussTedrake