RobotLocomotion/drake #1282: Differential flatness and feedback linearization

hjjayakrishnan, 2015-08-24:
Is differential flatness another word for feedback linearizable(with stable internal dynamics)?

In the case for the quadrotor, feedback linearization control is made possible by means of a dynamic extension to increase the relative degree inorder to attain stable internal dynamics. Differential flatness also means the same thing, right? You get to control all the states by just controlling x,y,z,psi and the rest of the states are stable by structure.

tkoolen, 2015-08-25:
They are not exactly the same thing, but there are connections.

See e.g. http://www.cds.caltech.edu/~murray/preprints/mrs95-imece.pdf, section 2.2.

Theorem 2 from that paper:
> If a control system is differentially flat then it is dynamic feedback linearizable on an open dense
set with the dynamic feedback possibly depending explicitly on time.

also,
> [...] chains of integrators and in fact all controllable linear or state feedback linearizable systems are differentially flat.

Note that this is not really my area of expertise though.

RussTedrake, 2015-08-27:
@tkoolen's reference is the right one.  They are not identical, but are morally very close.

closed 2015-08-27 by RussTedrake