RobotLocomotion/drake #1295: Pendulum.h assumes it's being run in the directory containing the urdf

rdeits, 2015-08-28:
The `PendulumWithBotVis` constructor tries to load "Pendulum.urdf". If the current executable isn't being run from that folder, it will throw a runtime_error and crash.

RussTedrake, 2015-08-28:
Right.  Thanks.  Fixed.

I’m not very happy with the idea that we’re going to start making lots of little executables, many specific to a particular robot, and install them all into pod-build/bin .  

What do people think about installing the binaries only (e.g. runLQR, runEnergyShaping, …) directly into the source directory?  it will mean you can’t cleanup by doing `rm pod-build` (but `make clean` will do the right thing).  But having the binaries only here will at least not cause any mysterious linking-to-a-library-i-didn’t-expect issues.  Plus the urdf’s etc are all in the source dir.

- Russ

rdeits, 2015-08-28:
Having in-source executables doesn't fix the pendulum path problem for me, because with the wrapped code it's the current directory of the python (or matlab) executable that matters. And I'm not all that fond of mixing up source and output files. 

If we don't want lots of robot-specific binaries on the path, what about a /bin/ folder for each example?

RussTedrake, 2015-08-28:
i thought about that, too.  examples/Pendulum/bin , etc?

psiorx, 2015-08-28:
I don't like the idea of installing the binaries alongside the source.  A separate bin folder per example would be better.

rdeits, 2015-08-28:
`examples/Pendulum/bin` sounds good to me

RussTedrake, 2015-08-28:
Re: hard-coded global paths.  It seems reasonable to me to have cmake set a global compiler flag defining DRAKE_PATH to be available (as ${PROJECT_SOURCE_DIR}) in all of our c code.  Then we could set the global path in the cpp file.  

Are there any reasons why this is morally wrong?

psiorx, 2015-08-28:
If you end up moving/copying the drake folder, things will break.

Wouldn't it be better to have everything relative to some anchor file that various other parts of the software can query for its location and base the path they are looking for on that?

RussTedrake, 2015-08-28:
we decided at some point that embracing the pods local install philosophy meant that we could write global paths to libraries, e.g. in the rpath.  that cleaned up a lot.  my point is that other things will also break already if you move/copy the drake folder.

RussTedrake, 2015-08-28:
another option is to take the urdf location as a command-line argument into the executable.  we'll have many apps, but could write a common argument parser.

rdeits, 2015-08-28:
Not having our classes assume a fixed URDF location seems like a good solution. It would mean that things like the `PendulumWithBotVis` constructor would need an additional argument giving the path to the URDF, but that's not necessarily a bad thing.

patmarion, 2015-08-30:
My 2 cents:

  - along with URDF files, are there other filesystem resources we should think about?
  - will you ever distribute the resources as part of a non-source installation (apt-get, homebrew)?  Resources are often installed to something like <install-prefix>/share/drake-<version>/

  - c++ api like PendulumWithBotVis should probably take args specifying the full path to files, and the calling code should determine where to find the files

  - when possible, convenience functions could be provided that makes it easy to get the path to required files, so most code doesn't have to worry about the problem. Something like:  DrakeResources::getExamplesBasePath()  or DrakeResources::getExampleUrdfFullPath('pendulum.urdf') if that's feasible

  - if you want to support a package install with resources, you might not want to hardcode the base path, but figure it out at runtime based on the location of the executable (or in Python, relative to the imported module).  But as an initial step, hardcoding based on ${PROJECT_SOURCE_DIR} seems fine.

  - instead of DRAKE_PATH as a compiler flag, consider putting the value into a configured header.  This could be a place to store other useful variables, like major/minor version, git sha1, etc.

RussTedrake, 2015-08-30:
> will you ever distribute the resources as part of a non-source installation (apt-get, homebrew)? Resources are often installed to something like /share/drake-/

Thanks Pat.  I appreciate the specific suggestions.

And Yes - we have many other resources that get distributed - i consider the .m files to be exactly that as well.

closed 2015-12-16 by rdeits