RobotLocomotion/drake #1296: Bug with writeMeshOBJ's use of vertex normals

mposa, 2015-08-28:
util/writeMeshOBJ.m seems to generate invalid OBJ files. In particular, the code to generate the normals ("vt x y z") looks wrong.

1) It only generates 2-vectors, not 3-vectors
2) The vectors it generates are unrelated to the actual normal computations

patmarion, 2015-09-28:
"vt" specifies texture coordinates, not vertex normals.  It appears correct, in that case.  It indeed looks like the code computes normals, but never writes them to the .obj file.  To write normals, it would use the "vn" specifier.

RussTedrake, 2015-12-27:
i think we've decided this is not an issue...

mposa, 2015-12-28:
Ok. I'll have to keep the fix around on my branch then, since it caused crashes for me.

RussTedrake, 2015-12-28:
can you tell us how to reproduce the crash?

mposa, 2016-01-06:
Seems to have been tied to the older visualizer

closed 2016-01-06 by mposa

RussTedrake, 2016-01-06:
great.