RobotLocomotion/drake #1300: looks like T_body_to_joint is not being used any more?

RussTedrake, 2015-08-29:
in matlab.  @tkoolen - can you confirm, and zap it if I'm correct?

RussTedrake, 2015-08-29:
(effectively you are recomputing it every time in jointTransform.m)

RussTedrake, 2015-12-27:
based on a quick search, it looks like it is still being used in changeRootLink and one of the BMI methods (but nowhere else).  could that be related to #417 ?

hongkai-dai, 2015-12-28:
Is `T_body_to_joint` fixed to identity? I can update the BMI method to get rid of this.

tkoolen, 2015-12-29:
`T_body_to_joint` should be removed.

`T_body_to_joint` is written to here:
* systems/plants/RigidBody.m
* systems/plants/@RigidBodyManipulator/changeRootLink.m
* systems/plants/@RigidBodyManipulator/RigidBodyManipulator.m

`T_body_to_joint` is read from here:
* solvers/BMI/kinematics/inverseKinematicsBMI.m
* solvers/BMI/kinematics/test/testKinematics.m
* systems/plants/RigidBodyTorsionalSpring.m
* systems/plants/@PlanarRigidBodyManipulator/PlanarRigidBodyManipulator.m

So `T_body_to_joint` can't be related to #417, as it's not used in any relevant places.

@hongkai-dai, I would appreciate it if you could get rid of it in the BMI code. `T_body_to_joint` has no meaning and any usages of it are bugs. Note that body.Ttree is the homogeneous transform from the predecessor frame of the joint associated with the body to the body's parent body frame, so that the transform from body frame to parent body frame is body.Ttree * jointTransform(body, q_body). There are no other transforms in between. So I guess you could assume that `T_body_to_joint` is always identity, yeah; but just get rid of any mention of it.

RussTedrake, 2015-12-29:
While I agree changeRootLink is only writing to T_body_to_joint, it could still be related to #417 if the changeRootLink method thinks it’s accomplishing a change to the kinematic tree which is not actually taking place.

hongkai-dai, 2015-12-29:
@tkoolen, thanks for the explanation. I think currently `T_body_to_joint` is not always identity yet. One example is
```
p = RigidBodyManipulator([getDrakePath,'/examples/IRB140/urdf/irb_140_robotiq_simple_ati.urdf'],struct('floating',true));
p.getBody(3).T_body_to_joint
```
And the homogeneous transformation is 
```
[1 0 0 0;
 0 0 1 0;
 0 -1 0 0;
 0 0 0 1]
```

tkoolen, 2015-12-30:
Does the non-identity `T_body_to_joint` cause issues in the BMI code after removing the lines that access `T_body_to_joint`?

hongkai-dai, 2015-12-30:
@tkoolen Got it, I should use body.joint_axis. The BMI code is fixed.

PR #1567 (2016-01-03): Remove RigidBody.T_body_to_joint and RigidBody.Xtree
Fixes #1300.
closed 2016-01-03 by RussTedrake