RobotLocomotion/drake #1307: BotVisualizer constructor should properly clear existing drake designer

tkoolen, 2015-08-31:
Currently if you run e.g. the unit tests in systems/plants/test, the visualizer will not erase some objects from tests that have already finished, e.g. these blue dots:

[image]

Is there a simple 'clear' message we could send in the constructor of BotVisualizer if a previous viewer was found?

patmarion, 2015-09-02:
Is it possible that those blue dots are from lcm-gl messages and not drake draw messages?

RussTedrake, 2016-05-01:
i would definitely suspect they are from lcmgl.  @tkoolen -- please let us know so we can act on it with @patmarion 's help this week if necessary.

jwnimmer-tri, 2017-06-19:
So the general problem here may still exist, but this particular realization of it is no longer relevant, both because these tests won't exist soon, and because lcmgl is not used very much within Drake anymore.

Closing as won't-fix, because #6369 will be removing the relevant software.

If I'm confused and this should in fact stay open, please re-open an clarify the remaining unsolved problem(s) in a new comment here.

closed 2017-06-19 by jwnimmer-tri