RobotLocomotion/drake #1308: Don't include installed (copied) headers

tkoolen, 2015-09-01:
Right now, lines 681 and 682 of pods.cmake include all headers in build/include and pod-build/include. This means that there are three copies of each header we pods_install in the header search path. I'm honestly amazed that this works. Despite the fact that it doesn't seem to break anything I find it quite annoying because CLion will open a different header when you click on an error depending on whether you've done a clean build or not.

What is the right solution for this? Can we include all directories in build/include *except* the one with the same name as the current pod? Can we get remove pod-build/include from the header search path altogether?

patmarion, 2015-09-02:
It does break things in DRC.  If we switch branches and drake moves, then it is possible to get compile errors due to included headers from the build directory.  And in this case it isn't fixable by make clean, my fix has been to find . -name *drake* | xargs rm -r in the build directory :(

rdeits, 2015-12-16:
Recommendation: use "before" property in include_directories in cmake to resolve ambiguities

RussTedrake, 2016-01-23:
this should be soundly resolved now.

closed 2016-01-23 by RussTedrake