RobotLocomotion/drake #1309: Fix "hack to force cmake always create install and clean targets"

tkoolen, 2015-09-01:
I'd been wondering for a while why randomly named run configurations kept on accumulating in CLion. I just found this in pods.cmake:
```
# hack to force cmake always create install and clean targets
install(FILES DESTINATION)
string(RANDOM LENGTH 32 __rand_target__name__)
add_custom_target(${__rand_target__name__})
unset(__rand_target__name__)
```
Could we please find a better solution?

RussTedrake, 2015-09-01:
Wow.  Never noticed that (it’s a carryover from the original pods.cmake).  

The wrapper make files no longer depend on their being an install target.
I’d be surprised if any of our projects fail to generate the clean target.

Have you tried simply removing it and seeing if anything breaks?

tkoolen, 2015-09-01:
I tried getting rid of these lines and I have no issues so far. I did some builds and cleans in different configurations. I'll submit a pull request to https://github.com/RobotLocomotion/cmake.

closed 2015-09-01 by RussTedrake