RobotLocomotion/drake #1311: Wrap our IK interface

rdeits, 2015-09-01:
I've got a basic demo of IK from python, including creating posture constraints and calling the solver, here: https://github.com/rdeits/drake/blob/rd-ik-python/drake/python/rbm.ipynb 

It should be straightforward to extend this to cover the entire old IK interface, but it will require some changes. Specifically, the constraint methods will need to have their arguments cleaned up to use Eigen vector types everywhere, instead of mixing Eigen types and C-style raw pointers. That won't be hard, but it will mean touching a lot of different code paths. 

We just have to decide if it's worth doing this cleanup on the old version of IK, or whether to just jump into  wrapping the new interface.

RussTedrake, 2015-09-01:
awesome. 

i’m working on the DrakeFunction bits today.  Should have a sense soon of how hard it will be to update the c++ to the new IK.

avalenzu, 2015-09-01:
> i’m working on the DrakeFunction bits today.

Exciting!

avalenzu, 2015-09-02:
On the other hand, with how well the AutoDiffScalars appear to be working drakeFunctions may be redundant in C++. In MATLAB they're getting us input/output size checks and proper composition of gradients. In C++, we could get that by using AutoDiffScalars and fixed size args and just use ... functions. :)

patmarion, 2015-09-02:
Director represents constraints in Python, so it could be a quick port to use this new python wrapping, if the old style IK interface is provided.

I haven't done a full review, but are there some features of the old IK interface that aren't available yet in the new style?

rdeits, 2016-04-01:
@patmarion and I have run into this again this week. Specifically, in order to wrap the existing C++ IK for python, the argument types need to be cleaned up (replacing bare pointers with std::vector types and Eigen types as appropriate).

RussTedrake, 2016-05-01:
the IK backend is being moved to the new c++ optimization classes.  i believe that should make this much easier.  (but I also think that you've already accomplished this).  @rdeits -- please close if possible.

rdeits, 2017-03-26:
this happened

closed 2017-03-26 by rdeits