RobotLocomotion/drake #1312: Eigen AutoDiff bug?

RussTedrake, 2015-09-01:
For some reason I can't compile this trivial code:
```
    Drake::TaylorVecX x(2);
    MatrixXd I = MatrixXd::Identity(2,2);
    Drake::TaylorVecX y = I*x;
```
I get all a huge list of errors, which look like it's just a bug in the Eigen headers (help/opinions appreciated!):
```
/Users/russt/drake-distro/build/include/eigen3/Eigen/src/Core/products/GeneralMatrixVector.h:79:62: error: implicit instantiation of undefined template 'Eigen::internal::conj_helper<double, Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, -1, 1> >, false, false>'
  conj_helper<LhsScalar,RhsScalar,ConjugateLhs,ConjugateRhs> cj;
                                                             ^
/Users/russt/drake-distro/build/include/eigen3/Eigen/src/Core/GeneralProduct.h:462:107: note: in instantiation of member function 'Eigen::internal::general_matrix_vector_product<long, double, 0, false, Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, -1, 1> >, false, 0>::run' requested here
      <Index,LhsScalar,ColMajor,LhsBlasTraits::NeedToConjugate,RhsScalar,RhsBlasTraits::NeedToConjugate>::run(
                                                                                                          ^
/Users/russt/drake-distro/build/include/eigen3/Eigen/src/Core/GeneralProduct.h:349:89: note: in instantiation of function template specialization 'Eigen::internal::gemv_selector<2, 0, true>::run<Eigen::GeneralProduct<Eigen::Matrix<double, -1, -1, 0, -1, -1>, Eigen::Matrix<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, -1, 1> >, -1, 1, 0, -1, 1>, 4>, Eigen::Matrix<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, -1, 1> >, -1, 1, 0, -1, 1> >' requested here
                       bool(internal::blas_traits<MatrixType>::HasUsableDirectAccess)>::run(*this, dst, alpha);
                                                                                        ^
/Users/russt/drake-distro/build/include/eigen3/Eigen/src/Core/ProductBase.h:118:81: note: in instantiation of function template specialization 'Eigen::GeneralProduct<Eigen::Matrix<double, -1, -1, 0, -1, -1>, Eigen::Matrix<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, -1, 1> >, -1, 1, 0, -1, 1>, 4>::scaleAndAddTo<Eigen::Matrix<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, -1, 1> >, -1, 1, 0, -1, 1> >' requested here
    inline void scaleAndAddTo(Dest& dst, const Scalar& alpha) const { derived().scaleAndAddTo(dst,alpha); }
                                                                                ^
/Users/russt/drake-distro/build/include/eigen3/Eigen/src/Core/ProductBase.h:109:58: note: in instantiation of function template specialization 'Eigen::ProductBase<Eigen::GeneralProduct<Eigen::Matrix<double, -1, -1, 0, -1, -1>, Eigen::Matrix<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, -1, 1> >, -1, 1, 0, -1, 1>, 4>, Eigen::Matrix<double, -1, -1, 0, -1, -1>, Eigen::Matrix<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, -1, 1> >, -1, 1, 0, -1, 1> >::scaleAndAddTo<Eigen::Matrix<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, -1, 1> >, -1, 1, 0, -1, 1> >' requested here
    inline void evalTo(Dest& dst) const { dst.setZero(); scaleAndAddTo(dst,Scalar(1)); }
                                                         ^
/Users/russt/drake-distro/build/include/eigen3/Eigen/src/Core/ProductBase.h:127:17: note: in instantiation of function template specialization 'Eigen::ProductBase<Eigen::GeneralProduct<Eigen::Matrix<double, -1, -1, 0, -1, -1>, Eigen::Matrix<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, -1, 1> >, -1, 1, 0, -1, 1>, 4>, Eigen::Matrix<double, -1, -1, 0, -1, -1>, Eigen::Matrix<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, -1, 1> >, -1, 1, 0, -1, 1> >::evalTo<Eigen::Matrix<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, -1, 1> >, -1, 1, 0, -1, 1> >' requested here
      derived().evalTo(m_result);
                ^
/Users/russt/drake-distro/drake/solvers/test/testSimpleDrakeFunctions.cpp:46:27: note: in instantiation of member function 'Eigen::ProductBase<Eigen::GeneralProduct<Eigen::Matrix<double, -1, -1, 0, -1, -1>, Eigen::Matrix<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, -1, 1> >, -1, 1, 0, -1, 1>, 4>, Eigen::Matrix<double, -1, -1, 0, -1, -1>, Eigen::Matrix<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, -1, 1> >, -1, 1, 0, -1, 1> >::operator const Eigen::Matrix<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, -1, 1> >, -1, 1, 0, -1, 1> &' requested here
    Drake::TaylorVecX y = I*x;
                          ^
/Users/russt/drake-distro/build/include/eigen3/Eigen/src/Core/util/ForwardDeclarations.h:151:97: note: template is declared here
template<typename LhsScalar, typename RhsScalar, bool ConjLhs=false, bool ConjRhs=false> struct conj_helper;
                                                                                                ^
In file included from /Users/russt/drake-distro/drake/solvers/test/testSimpleDrakeFunctions.cpp:2:
In file included from /Users/russt/drake-distro/drake/solvers/test/../DrakeFunction.h:7:
In file included from /Users/russt/drake-distro/drake/util/drakeGradientUtil.h:4:
In file included from /Users/russt/drake-distro/build/include/eigen3/Eigen/Core:329:
/Users/russt/drake-distro/build/include/eigen3/Eigen/src/Core/products/GeneralMatrixVector.h:80:62: error: implicit instantiation of undefined template 'Eigen::internal::conj_helper<double, Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, -1, 1> >, false, false>'
  conj_helper<LhsPacket,RhsPacket,ConjugateLhs,ConjugateRhs> pcj;
                                                             ^
/Users/russt/drake-distro/build/include/eigen3/Eigen/src/Core/util/ForwardDeclarations.h:151:97: note: template is declared here
template<typename LhsScalar, typename RhsScalar, bool ConjLhs=false, bool ConjRhs=false> struct conj_helper;
                                                                                                ^
2 errors generated.
```

RussTedrake, 2015-09-01:
replacing the type in my last line with `auto` resolves it, but the actual use case that I care about is defining a method which must declare a return type.  What's the right declaration for that return?

psiorx, 2015-09-01:
You can use the trailing return type syntax potentially.  Something like:

```C++
myfunction(Eigen::MatrixXd I, Drake::TaylorVecX x) -> decltype(I * x)
```

RussTedrake, 2015-09-01:
Cool trick.  But I’m trying to build the class hierarchy around virtual methods with a known return type.  So I’d prefer to force Eigen to give me back some reasonable return type that will work for all of our functions.

psiorx, 2015-09-01:
Not sure what type it's actually resolving to.  You could try this:

```C++
#include <typeinfo>
...
std::cout << typeid(I * x).name() << std::endl;

```

To print the runtime type and see if that reveals anything.

RussTedrake, 2015-09-01:
It’s in the error message (the GeneralProduct).  But I really think it’s a bug in eigen’s headers because I should be able to copy / assign it to the normal type.

tkoolen, 2015-09-02:
I'm not 100% sure this is considered a bug, since Eigen doesn't allow mixing different scalar types in matrix expressions anywhere else. Normally it prints out a more helpful message though. As a workaround, try this:
```
  Drake::TaylorVecX x(2);
  MatrixXd I = MatrixXd::Identity(2,2);
  Drake::TaylorVecX y = I.cast<typename Drake::TaylorVecX::Scalar>() * x;
```

RussTedrake, 2015-09-02:
What?  I totally disagree... :). This must happen all of the time in the other autodiff code (multiplying taylorvars by non-taylorvar matrices).  And actually that part works fine -- what fails is the conversion at the end back to a standard taylorvar (from the product blah)

I think i can get around it by defining my own converter locally

tkoolen, 2015-09-02:
I did have to cast explicitly in a limited number of cases (in all cases, operations involving fixed parameters such as inertias, the pitch of a helical joint, ...). It wasn't terrible in the case of RBM. But I agree that it would be nice to have this work without doing anything special.

RussTedrake, 2016-01-17:
ran into this again, and found http://eigen.tuxfamily.org/bz/show_bug.cgi?id=279 , where i think it's fair to say that they acknowledge the issue.  the explicit casts you're doing are presumably inefficient, and in my view should not be required.

RussTedrake, 2016-05-01:
@amcastro-tri -- this is the one I was referring to.

RussTedrake, 2016-05-01:
@ggould-tri, could you take a look?  fixing this soon could save us from writing a lot of bad c++ code to support autodiff.  if fixed, we should also zap the existing bad code, like this explicit casts required throughout the constraint classes (e.g., https://github.com/RobotLocomotion/drake/blob/master/drake/solvers/Constraint.h#L123 ).  that also appears in some of the rigid body tree methods.

amcastro-tri, 2016-05-02:
Thank you @RussTedrake! we were discussion this issue in particular with @david-german-tri and @tkoolen on Friday for the new joints design that'd allow automatic differentiation on parameters. Eigen was originally designed for performance on floating point operations. With their API they force the user to use the same Scalar type in matrices being part of an operation. This is to avoid for instance the multiplication of a <float> matrix with a <double> matrix. Such an operation requires casting and Eigne's API requires the user to explicitly call the `cast<ToScalar>` method in order to explicitly make aware the user of this casting. 
Therefore this check for both types being the same is buried well within Eigen's internals. Not sure if it can be hacked, or even if we want to....

david-german-tri, 2016-05-02:
The design consensus from the conversation @amcastro-tri mentioned has been written up: [public read-only with 5min latency](https://docs.google.com/a/tri.global/document/d/16fxMgXaqdM0Raa1UbAvE8NpozzdCBcgx9egJjM6Rr94/pub), [editable](https://docs.google.com/document/d/16fxMgXaqdM0Raa1UbAvE8NpozzdCBcgx9egJjM6Rr94/edit#heading=h.9ztfskti2res)

amcastro-tri, 2016-05-02:
@sherm1, I would love your input about this document @david-german-tri wrote. In the design we are proposing joints are templated on the Scalar (which can be say double, AutoDiff, Poly, etc). We decided to do so because as @RussTedrake explained it to me AutoDiff scalars are pretty vicious things. In order for the auto differentiation to work "everything" must be an AutoDiff or otherwise you run into operations between different types that can only be performed if converting (promoting in #2209) to AutoDiff. This is why we though: "might as well just have everything in the class to be an AutoDiff from the get go". This would help clean up #2209 quite a bit and allow having one abstract "Joint" class from which all other joints can be derived. 
Disadvantages (if any): if the user need both simulation (all performed with doubles) and auto differentiation of the same RBT, then that would require having two RBT's: `RBT<double>` and `RBT<AutodDiff>`, two different types.

amcastro-tri, 2016-05-02:
There is a [draft to our first design](https://docs.google.com/document/d/1Oo_gmiD8ZMIcNHN5YnGxYEC-oD3UybGqxJA_pkiQNo8/edit#).

sherm1, 2016-05-02:
I am totally in favor of the templatize-everything-by-scalar approach, so that we can have RBT<double> and RBT<AutoDiff> -- that's very clean and would allow other derivative-calculating scalar types, such as my favorite, [complex step derivatives](http://biomedicalcomputationreview.org/2/3/8.pdf).

But ... I think we have to do something about the mapping of variables-of-interest to the entries in the AutoDiff vector. Maybe I'm misunderstanding, but that looks near-impossible to get right for a large system with zillions of state variables and design parameters.

tkoolen, 2016-05-02:
Is this the right place to discuss the design document? (btw, I lost access to it)

sherm1, 2016-05-02:
There seem to be two different documents? (Alejandro's and David's links go to different places.) And I don't have access, not even commenting ability on one of them.

david-german-tri, 2016-05-02:
The logistics have gone pear-shaped!  Three things:

1. There are two documents, [one written by me mostly in English](https://docs.google.com/document/d/16fxMgXaqdM0Raa1UbAvE8NpozzdCBcgx9egJjM6Rr94/edit#heading=h.9ztfskti2res) and [one written by Alejandro mostly in C++](https://docs.google.com/document/d/1Oo_gmiD8ZMIcNHN5YnGxYEC-oD3UybGqxJA_pkiQNo8/edit#heading=h.t701r2n66v4r).  I've copied Alejandro's code into my doc.  Let's iterate there going forward.

2. @sherm1 You should now have comment access to the unified doc.  However, sharing Google Docs from tri.global to contributors outside TRI broke over the weekend.  I'm trying to get it fixed.  If I can't, I'll just move the document off the TRI domain altogether.

3. This issue is probably the wrong place to discuss Joints 2.0.  Let's move to #2235.

ggould-tri, 2016-05-25:
Per comment at https://github.com/RobotLocomotion/drake/issues/1312#issuecomment-216009861 I am incrementing the priority here.

amcastro-tri, 2016-05-25:
This does't seem a trivial problem (at least to me). @ggould-tri, so you just get the flavor of it, try multiplying a matrix of `float` by a matrix of `double`. Eigen immediately complaints a about this from its very deep internals (there'll be a static check that ends on your screen with words in capital letters separated by underscores).

tkoolen, 2016-05-26:
I addressed this problem in my first version of the DrakeJoint pull request, https://github.com/RobotLocomotion/drake/pull/2209

See `Promote` in drake/util/TypeConversions.h in that pull request.

RussTedrake, 2016-11-27:
i believe that this issue has been resolved in the newer versions of Eigen.  There might still be explicit casts sprinkled through the code (especially in RBT, @amcastro-tri) that could now be removed.  But I think it's safe to close this issue.

closed 2016-11-27 by RussTedrake