RobotLocomotion/drake #1317: Purge the old controller

rdeits, 2015-09-02:
I'd like to get rid of the old QPController implementation since it's no longer used on the robot or in any simulation examples except the running and jumping demos. To do that, I'll need to:

- [ ] extend lcmt_zmp_data to support 3D systems
- [ ] generalize QPLocomotionPlan.cpp to accept non-constant S in the Lyapunov function
- [ ] generalize setupAndSolveQP to variable-size linear systems (for 2D and 3D)

RussTedrake, 2016-01-23:
@rdeits -- did the push for valkyrie balancing accomplish this?

rdeits, 2016-01-26:
No, unfortunately. It's an unrelated issue, mostly due to the lack of support for 3D linear systems in the new controller.

jwnimmer-tri, 2017-06-19:
Even though #6369 is going to remove the MATLAB implementation, my guess is that it makes sense to keep this issue open, as a TODO to work on the C++ implementation.

RussTedrake, 2018-11-11:
related to #6381.  with that link established, I will close this one.

closed 2018-11-11 by RussTedrake