RobotLocomotion/drake #1319: Error building make from GitShell

itsamk, 2015-09-05:
It gives me the following error when I try to build drake on my system using GitShell.[See the image] Please help me resolve this. Thank you. I have already spent hours on the installation.
[image]

RussTedrake, 2015-09-05:
I believe this is because your make command has a space in it.  But this should have already been resolved in the master branch of drake.  Can you make sure you are up to date, and show me a git log to be sure?

itsamk, 2015-09-05:
I am not really aware with what a git-log is. Can you specifically tell me what to do? I have followed all the instructions on the drake installation instructions page.

RussTedrake, 2015-09-05:
in the shell, cd into drake-distro and type `git log`.

itsamk, 2015-09-05:
I exported the git log to a text file and that gave me a 3.7 MB sized file that I can't upload here. Any other way?

RussTedrake, 2015-09-05:
I just need the top 10 lines

RussTedrake, 2015-09-05:
I'm planning to make a new precompiled distribution in the next few days -- probably even this weekend.  Perhaps it's easier to go the precompiled route?

itsamk, 2015-09-05:
Here is the head of the text file. See if you can figure out the problem. It will be really great if this could work somehow now. Otherwise, I can wait till you build up a precompiled version. Sooner will be the better. 
------------------------------ 
commit d42493ab9f1753ace73f9d47851ff76ba2f54061
Merge: 9142442 f381eaf
Author: Russ Tedrake <russt@mit.edu>
Date:   Thu Sep 3 05:22:51 2015 -0400

    Merge pull request #1316 from rdeits/rd-controller-cleanup
    
    delete a bunch of unused control files

commit 91424424785c7ec8b95872d58e426871d6a70482
Merge: 178dd45 775f19b
Author: Russ Tedrake <russt@mit.edu>
Date:   Thu Sep 3 04:44:56 2015 -0400

    Merge pull request #1313 from tkoolen/tk-update-cmake-remove-random-target
    
    Update cmake (got rid of random target generation in pods.cmake)
------------------------------------------

RussTedrake, 2015-09-05:
took a quick look.  the short answer is -- don't use GitShell (it doesn't support batch syntax).  try running make in the cmd.exe window instead.  I've also used Git Bash and or Git CMD.

For my own notes: PowerShell should probably be treated as Cygwin instead of Windows_NT, but it still seems to have handled some of the quotes differently.

itsamk, 2015-09-05:
Thanks for the suggestion. 'Make' finally executes but gets stuck somewhere in the process.[see the image] 
[image]

RussTedrake, 2015-09-05:
git is asking you for a password.  snopt requires a license.  you’ll have to set WITH_SNOPT to false for now.

itsamk, 2015-09-05:
Gives me this error now. 
[image]

RussTedrake, 2015-09-05:
the actual error is higher up in the screen.  search for the word Error.

itsamk, 2015-09-05:
[image]

Here.

RussTedrake, 2015-09-05:
no.  i need to know what cmd.exe was called when it failed.

itsamk, 2015-09-05:
Sorry, I realized that. I updated the previous comment.

RussTedrake, 2015-09-05:
it looks like drake did not get checked out properly.  perhaps because it was partially checked out during one of your previous failed attempts.  i would do a clean build (and set your option preferences again)
```
cd drake-distro
rd /s build
rd /s pod-build
make options
make
```

itsamk, 2015-09-05:
[image]

Back to the previous error.

RussTedrake, 2015-09-05:
what happens if you cd in the drake subdirectory and run make in there?

itsamk, 2015-09-05:
[image]

gives me this.

RussTedrake, 2015-09-05:
you haven’t put the pkg config lite program in your path (from the installation instructions)

itsamk, 2015-09-05:
[image]

Here, right? I did.

RussTedrake, 2015-09-05:
can you call pkg-config from your shell?   (you have to restart the shell for environment variable changes to take effect)

itsamk, 2015-09-05:
No. It says the term is not recognized.

RussTedrake, 2015-09-05:
you need to resolve that.

itsamk, 2015-09-05:
I have added it in the path so theoretically it should work right? 
[..]\Downloads\pkg-config-lite-0.28-1_bin-win32\pkg-config-lite-0.28-1\bin;

cmd.exe gives me the same error.
git shell gives me this:
[image]

RussTedrake, 2015-09-05:
that’s working.  try make again

itsamk, 2015-09-05:
The error is still there. I have even set WITH_EIGEN and WITH_SPOTLESS to FALSE.(They gave errors too)
[image]

RussTedrake, 2015-09-05:
You need eigen.  it’s a required dependency.  

I’m sorry — but I think you should use the precompiled distro.

itsamk, 2015-09-05:
Then, if I turn EIGEN to TRUE it still gives be related errors. I wish there was some resolution to this. I have just started the course "Underactuated Robotics" at edz and I hope the lack of drake for the first week doesn't cause much problems.
Waiting for the precompiled version. Thanks for your time.

itsamk, 2015-09-06:
I just realized that you're the instructor to the course as well. Thanks a lot for all the great material. Means a lot.

itsamk, 2015-09-15:
I removed drake and did a fresh start and cloned it again. This time I used gitbash to run make but, somehow make is giving me an error. Please help me through this.
[image]

RussTedrake, 2015-09-15:
did make options work?  did it pull up a gui, and let you configure and generate?

itsamk, 2015-09-15:
Yes, it did!

RussTedrake, 2015-09-15:
try running `make -d` instead of just `make` and show us the output?

itsamk, 2015-09-15:
[image]

RussTedrake, 2015-09-15:
i see.  i think that cmake did not find your make executable.  to confirm, can you check the file pod-build/CMakeCache.txt for the MAKE_EXECUTABLE line and show us what it says?

and type `which make` at the command line?

itsamk, 2015-09-15:
I think I see the error. It has the wrong path set. What should be the correct path here?
[image]

RussTedrake, 2015-09-15:
wherever you have make installed.

itsamk, 2015-09-15:
Changed it to MAKE_EXECUTABLE:FILEPATH= C:/Program Files (x86)/GnuWin32/bin/make.exe
still doesnt work :\

RussTedrake, 2015-09-15:
make sure there are no extra spaces and put quotes around it?

itsamk, 2015-09-15:
I double checked that.

itsamk, 2015-09-15:
Tried quotes too.

tkoolen, 2015-09-21:
So https://github.com/RobotLocomotion/drake/pull/1353 fixed this right?

RussTedrake, 2015-09-21:
yes.

closed 2015-09-21 by RussTedrake