RobotLocomotion/drake #1325: separate configure and build steps for gnu make pods externals?

RussTedrake, 2015-09-08:
Just noticed that one result of us calling make exactly once is that we pass in the BUILD_TYPE for both the configure and build steps.  On msvc, that results in project files being built which support multiple build types, but are all hard-coded with the types passed in a configure time:

```
if %errorlevel% neq 0 goto :VCEnd</Command>
      <AdditionalInputs Condition="'$(Configuration)|$(Platform)'=='Debug|x64'">Z:/russt/drake-distro-win/pod-build/CMakeFiles/04423273ec4096187fbff0ee69951c62/gtk-build.rule;Z:\russt\drake-distro-win\pod-build\gtk-prefix\src\gtk-stamp\$(Configuration)\gtk-configure;%(AdditionalInputs)</AdditionalInputs>
      <Outputs Condition="'$(Configuration)|$(Platform)'=='Debug|x64'">Z:\russt\drake-distro-win\pod-build\gtk-prefix\src\gtk-stamp\$(Configuration)\gtk-build</Outputs>
      <LinkObjects Condition="'$(Configuration)|$(Platform)'=='Debug|x64'">false</LinkObjects>
      <Message Condition="'$(Configuration)|$(Platform)'=='Release|x64'">Performing build step for 'gtk'</Message>
      <Command Condition="'$(Configuration)|$(Platform)'=='Release|x64'">setlocal
cd Z:\russt\drake-distro-win\externals\gtk
if %errorlevel% neq 0 goto :cmEnd
Z:
if %errorlevel% neq 0 goto :cmEnd
"C:\Program Files (x86)\GnuWin32\bin\make.exe" BUILD_PREFIX=Z:/russt/drake-distro-win/build BUILD_TYPE=Release
if %errorlevel% neq 0 goto :cmEnd
"C:\Program Files (x86)\CMake\bin\cmake.exe" -E touch Z:/russt/drake-distro-win/pod-build/gtk-prefix/src/gtk-stamp/$(Configuration)/gtk-build
if %errorlevel% neq 0 goto :cmEnd
:cmEnd
endlocal &amp; call :cmErrorLevel %errorlevel% &amp; goto :cmDone
:cmErrorLevel
exit /b %1
:cmDone
if %errorlevel% neq 0 goto :VCEnd</Command>
      <AdditionalInputs Condition="'$(Configuration)|$(Platform)'=='Release|x64'">Z:/russt/drake-distro-win/pod-build/CMakeFiles/04423273ec4096187fbff0ee69951c62/gtk-build.rule;Z:\russt\drake-distro-win\pod-build\gtk-prefix\src\gtk-stamp\$(Configuration)\gtk-configure;%(AdditionalInputs)</AdditionalInputs>
      <Outputs Condition="'$(Configuration)|$(Platform)'=='Release|x64'">Z:\russt\drake-distro-win\pod-build\gtk-prefix\src\gtk-stamp\$(Configuration)\gtk-build</Outputs>
      <LinkObjects Condition="'$(Configuration)|$(Platform)'=='Release|x64'">false</LinkObjects>
      <Message Condition="'$(Configuration)|$(Platform)'=='MinSizeRel|x64'">Performing build step for 'gtk'</Message>
      <Command Condition="'$(Configuration)|$(Platform)'=='MinSizeRel|x64'">setlocal
cd Z:\russt\drake-distro-win\externals\gtk
if %errorlevel% neq 0 goto :cmEnd
Z:
if %errorlevel% neq 0 goto :cmEnd
"C:\Program Files (x86)\GnuWin32\bin\make.exe" BUILD_PREFIX=Z:/russt/drake-distro-win/build BUILD_TYPE=Release
if %errorlevel% neq 0 goto :cmEnd
"C:\Program Files (x86)\CMake\bin\cmake.exe" -E touch Z:/russt/drake-distro-win/pod-build/gtk-prefix/src/gtk-stamp/$(Configuration)/gtk-build
if %errorlevel% neq 0 goto :cmEnd
:cmEnd
endlocal &amp; call :cmErrorLevel %errorlevel% &amp; goto :cmDone
:cmErrorLevel
exit /b %1
:cmDone
if %errorlevel% neq 0 goto :VCEnd</Command>
```

RussTedrake, 2015-12-15:
another thing to discuss in #1523

RussTedrake, 2015-12-27:
i believe this was resolved with #1538

tkoolen, 2016-01-28:
Unfortunately, this has not been resolved. I still can't switch configurations on the fly from within visual studio.

RussTedrake, 2016-05-01:
now i believe it has been resolved.  @sherm1 -- please correct me if I'm wrong

closed 2016-05-01 by RussTedrake

sherm1, 2016-05-01:
@RussTedrake: Not fixed yet but adequately covered by #2094 so this can stay closed.