RobotLocomotion/drake #1327: simulate() fails for all systems in Matlab 2015b

rdeits, 2015-09-09:
I just updated my mac to matlab r2015b, and I can no longer run any of the simulation examples. It looks like Simulink changed their API:

```
Error using getInitialState
Using S-Function API to get output or state in 'Dataset' format is currently not supported. The model must be
recompiled with ''Configuration Parameters > Data Import/Export > Format'' parameter set to 'Array',
'Structure' or 'Structure with time'.

Error in sl (line 15)
[varargout{1:nargout}]=feval(varargin{:});

Error in Simulink.BlockDiagram.getInitialState

Error in getSimulinkStateStructure (line 6)
x_struct = Simulink.BlockDiagram.getInitialState(mdl);

Error in DynamicalSystem/stateVectorToStructure (line 655)
        obj.structured_x = getSimulinkStateStructure(mdl.name);

Error in DynamicalSystem/simulate (line 69)
  x0 = stateVectorToStructure(obj,x0,mdl);

Error in atlasUtil.simulateWalking (line 32)
ytraj = simulate(sys, [0, T], x0, options);

Error in runAtlasWalking (line 68)
[ytraj, com, rms_com] = atlasUtil.simulateWalking(r, walking_plan_data);
```

rdeits, 2015-09-09:
too tired to figure this out right now, but I'll look at it tomorrow

RussTedrake, 2015-09-09:
Awesome.  Will take a look.  - Russ

PR #1328 (2015-09-10): set data export parameter for 2015b compliance
resolves #1327 
let's hope it also works for the other matlab versions/platforms.
closed 2015-09-10 by RussTedrake