RobotLocomotion/drake #1332: Creating Monitor causes matlab crash

mauricefallon, 2015-09-11:
One of our students is using a machine with Ubuntu 14.04 with Matlab2015a.
Drake & Matlab repeatedly crashes during the compile step when creating a monitor for the footstep planner, e.g. RobotStateMonitor()

Any idea what might be the source of this? Have these monitors been tested with Matlab2015a?
The crash dump and launch sequence is below.

$ matlab 
$ p = BasePlanner()

(process:1787): GLib-WARNING **: unknown option bit(s) set
lcm_subscribe: Error while compiling regular expression ^LCM_SELF_TEST$ at char 0: unknown option bit(s) set

(process:1787): GLib-WARNING **: unknown option bit(s) set
lcm_subscribe: Error while compiling regular expression ^LCM_SELF_TEST$ at char 0: unknown option bit(s) set
LCM self test failed!!
Check your routing tables and firewall settings
*** buffer overflow detected ***: /usr/local/MATLAB/R2015a/bin/glnxa64/MATLAB terminated
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x7338f)[0x7f0257d6238f]
/lib/x86_64-linux-gnu/libc.so.6(__fortify_fail+0x5c)[0x7f0257df9c9c]
/lib/x86_64-linux-gnu/libc.so.6(+0x109b60)[0x7f0257df8b60]
/lib/x86_64-linux-gnu/libc.so.6(+0x10abe7)[0x7f0257df9be7]
/home/snobili/ipab-distro/drc/software/build/matlab/../matlab/RobotStateMonitor.mexa64(_Z13lcmThreadMainv+0x92)[0x7f00ee27dc06]
/home/snobili/ipab-distro/drc/software/build/matlab/../matlab/RobotStateMonitor.mexa64(_ZNSt6thread5_ImplISt12_Bind_simpleIFPFvvEvEEE6_M_runEv+0x7)[0x7f00ee27f3b5]
/usr/local/MATLAB/R2015a/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6(+0xb1a40)[0x7f025889ea40]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x8182)[0x7f02580bc182]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x6d)[0x7f0257de947d]
======= Memory map: ========
00400000-0040a000 r-xp 00000000 08:01 4085658                            /usr/local/MATLAB/R2015a/bin/glnxa64/MATLAB
00609000-0060a000 rw-p 00009000 08:01 4085658                            /usr/local/MATLAB/R2015a/bin/glnxa64/MATLAB
00daf000-00fe0000 rw-p 00000000 00:00 0                                  [heap]
41687000-41723000 rw-p 00000000 00:00 0 
41a5c000-41a5e000 r-xs 00000000 08:01 5511326                            /tmp/.glU3r5Dy (deleted)
e0000000-e7a80000 rw-p 00000000 00:00 0 
e7a80000-e8000000 rw-p 00000000 00:00 0 
e8000000-f3d00000 rw-p 00000000 00:00 0 
f3d00000-fa000000 rw-p 00000000 00:00 0 
fa000000-ff900000 rw-p 00000000 00:00 0 
ff900000-100000000 ---p 00000000 00:00 0 
7f00eb7a1000-7f00eb7a2000 ---p 00000000 00:00 0 

------------------------------------------------------------------------
              abort() detected at Fri Sep 11 09:08:55 2015
------------------------------------------------------------------------

Configuration:
  Crash Decoding      : Disabled
  Crash Mode          : continue (default)
  Current Graphics Driver: Unknown hardware 
  Current Visual      : 0x21 (class 4, depth 24)
  Default Encoding    : UTF-8
  GNU C Library       : 2.19 stable
  Host Name           : daisy
  MATLAB Architecture : glnxa64
  MATLAB Root         : /usr/local/MATLAB/R2015a
  MATLAB Version      : 8.5.0.197613 (R2015a)
  OpenGL              : hardware
  Operating System    : Linux 3.13.0-58-generic #97-Ubuntu SMP Wed Jul 8 02:56:15 UTC 2015 x86_64
  Processor ID        : x86 Family 6 Model 60 Stepping 3, GenuineIntel
  Virtual Machine     : Java 1.7.0_60-b19 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
  Window System       : The X.Org Foundation (11501000), display :0

Fault Count: 1

Abnormal termination:
abort()

Register State (from fault):
  RAX = 0000000000000000  RBX = 0000000000000059
  RCX = ffffffffffffffff  RDX = 0000000000000006
  RSP = 00007f0e89a1ea28  RBP = 00007f0e89a1ed40
  RSI = 0000000000004c58  RDI = 0000000000004b18

   R8 = 0000000000000000   R9 = 00007f0ff3651927
  R10 = 0000000000000008  R11 = 0000000000000206
  R12 = 00007f0e89a1ebb0  R13 = 0000000000000005
  R14 = 0000000000000059  R15 = 0000000000000005

  RIP = 00007f0ff35e4cc9  EFL = 0000000000000206

   CS = 0033   FS = 0000   GS = 0000

Stack Trace (from fault):
[  0] 0x00007f0ff35e4cc9                    /lib/x86_64-linux-gnu/libc.so.6+00224457 gsignal+00000057
[  1] 0x00007f0ff35e80d8                    /lib/x86_64-linux-gnu/libc.so.6+00237784 abort+00000328
[  2] 0x00007f0ff3621394                    /lib/x86_64-linux-gnu/libc.so.6+00471956
[  3] 0x00007f0ff36b8c9c                    /lib/x86_64-linux-gnu/libc.so.6+01092764 __fortify_fail+00000092
[  4] 0x00007f0ff36b7b60                    /lib/x86_64-linux-gnu/libc.so.6+01088352
[  5] 0x00007f0ff36b8be7                    /lib/x86_64-linux-gnu/libc.so.6+01092583
[  6] 0x00007f0e89c42c06 /home/snobili/ipab-distro/drc/software/build/matlab/../matlab/RobotStateMonitor.mexa64+00019462 _Z13lcmThreadMainv+00000146
[  7] 0x00007f0e89c443b5 /home/snobili/ipab-distro/drc/software/build/matlab/../matlab/RobotStateMonitor.mexa64+00025525 _ZNSt6thread5_ImplISt12_Bind_simpleIFPFvvEvEEE6_M_runEv+00000007
[  8] 0x00007f0ff415da40 /usr/local/MATLAB/R2015a/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6+00727616
[  9] 0x00007f0ff397b182              /lib/x86_64-linux-gnu/libpthread.so.0+00033154
[ 10] 0x00007f0ff36a847d                    /lib/x86_64-linux-gnu/libc.so.6+01025149 clone+00000109
[ 11] 0x0000000000000000                                   <unknown-module>+00000000

RussTedrake, 2015-09-11:
> [ 11] 0x0000000000000000 +00000000

This is the known issue since the drc and was the reason that we pegged the matlab version.  It was specific to the drc codebase.  It definitely needs to be resolved.

RussTedrake, 2015-09-11:
the cause (and some initial discussion on a solution) are all documented here:
https://github.com/mitdrc/drc/issues/644

long story short, matlab includes two libraries in their library path that have the same symbols (one from pcre and the other from poco).  lcm links against the one and then, since 2014b gets called with the other.

mauricefallon, 2015-09-11:
Thanks Russ. I'll step back to 2014a.

closed 2015-09-11 by mauricefallon