RobotLocomotion/drake #1338: it seems not all cmake externals support "make clean"

RussTedrake, 2015-09-13:
maybe, at the very least, I could use http://www.cmake.org/cmake/help/v2.8.8/cmake.html#command%3aset_directory_properties to add the pod-build directory to make clean for subdirs...?

patmarion, 2015-09-29:
Probably already aware, but just to be clear, "make clean" means different thing in the cmake world vs. the pod world.

cmake's make clean: removes files created during the build (.o, .so, etc).  Does not remove configuration (CMakeCache.txt) and does not remove files created during install.

pods's make clean: rm's files listed in the install_manifest.txt (usually this lists all files that were installed, no guarantees for custom install steps), then rm's the pod-build (configuration data).  This is more like a make uninstall and make confclean.

jhoare, 2016-03-04:
Once this gets fixed up, I propose that we add a test in continuous integration to test ``make clean``  @jamiesnape

jamiesnape, 2016-03-04:
Create a ticket if others agree.

Lucy-tri, 2016-03-15:
I am working on creating a new external, nlopt #1627 , and it would be really handy to be able to reliably "make clean" for externals (and for everything else).

PR #1901 (2016-03-21): Better make clean support.
This improves the make clean support for the drake superbuild.

``make clean`` will still fail on fresh checkout, but ``make && make clean`` will finish successfully. 

I believe this Fixes #1338
closed 2016-03-21 by RussTedrake