RobotLocomotion/drake #1340: runAtlasWalkingNarrowStairs errors

RussTedrake, 2015-09-13:
on some of the build machines... ?  is that just because it was timing out on the others?  is there stochasticity in there?

http://drake006.csail.mit.edu/drake/pull-RussTedrake-snopt_alloc_on_heap-into-RobotLocomotion-master_20150913-0625-Continuous

<matlab_test>strcmp(test_name,'examples/Atlas/test/runAtlasWalkingNarrowStairs') && strcmp(error_id,'Drake:runAtlasWalkingStairs:BadCoMTracking')</matlab_test>

rdeits, 2015-09-14:
Nope, no stochasticity in that test (but that doesn't mean the results are reliable). I'll look at it now.

rdeits, 2015-09-14:
Have we maybe lost the fix that improved raycast accuracy in bullet? The raycast terrain height maps for that example are pretty gross now: 

[image]

psiorx, 2015-09-14:
This might have something to do with the recent bullet upgrade.  The patch that fixed the raycast problems was no longer relevant since the changes that it made are now the defaults in bullet.

rdeits, 2015-09-14:
Ok, good to know. It's not a critical issue, but might be worth remembering in case we have other raycast problems in the future.

PR #1344 (2015-09-15): slightly increase tolerance for narrow stairs test
fixes #1340 

execution still looks great, we were just slightly outside the (pretty narrow) tolerances.
commit b4a3f78e (2015-09-15): slightly increase tolerance for narrow stairs test

fixes #1340
closed 2015-09-15 by RussTedrake