RobotLocomotion/drake #1356: New rm + rd clean is not quiet

tkoolen, 2015-09-21:
If I do for example `make clean-spotless` on OSX, I see
```
make[5]: rd: No such file or directory
make[5]: rd: No such file or directory
```
Can we move the removal of build and pod-build into the CMakeLists.txt and use e.g. additional_make_clean_files or just file(REMOVE [file1 ...]) instead?

tkoolen, 2015-09-23:
I came up with a solution that makes CMake take care of the platform independence for us:

Instead of the current rm / rd approach, use
cmake -E remove_directory pod-build

For good measure, instead of @-mkdir pod-build, I would propose using
cmake -E make_directory pod-build

Wasn't aware of the -E option (CMake command mode) until now.

@RussTedrake, you wouldn't happen to have a one shot system for updating all of the Makefiles for all of our projects?

RussTedrake, 2015-09-23:
good call.  i like this solution.  (and should have thought of it!  i’m using the command mode all over the place now… :)

RussTedrake, 2015-09-23:
and no, don't have a magic button, but am getting fast at it.  pull request coming.

PR #1363 (2015-09-23): build logic fixes
resolves #1356
also resolves the WITH_ALL_SUPPORTED_EXTERNALS logic error reported in #1361
commit 3bca3ea5 (2015-09-23): resolves #1356
also resolves the WITH_ALL_SUPPORTED_EXTERNALS logic error reported in #1361
closed 2015-09-23 by RussTedrake