RobotLocomotion/drake #1365: RoA examples are failing with "Problem looks primal infeasible"

rdeits, 2015-09-23:
On my system, I'm getting errors like this: 

```
Error using PolynomialSystem/regionOfAttraction>levelSetMethod (line 334)
Problem looks primal infeasible

Error in PolynomialSystem/regionOfAttraction (line 136)
      V = levelSetMethod(V,f,options);

Error in CartPolePlant/balanceLQR (line 62)
        V=regionOfAttraction(pp,V,options);

Error in runLQR (line 8)
  [c,V] = balanceLQR(d);
```

for the `runLQR` demos in Quadrotor2D and CartPole. I'm also getting this error on some of the textbook examples like `testCubicPolynomialRoa`.

psiorx, 2015-09-23:
@anirudhamajumdar 
I think these started cropping up after enabling input saturations.  Any idea what could be going wrong?

rdeits, 2015-10-08:
The issues seems to be only with Mosek. Forcing options.solver to be `@spot_sedumi` fixes it for me

rdeits, 2015-10-08:
Turns out that's just the error that spotless reports when your mosek license has expired.

closed 2015-11-24 by psiorx