RobotLocomotion/drake #1368: make clean-lcm fails

tkoolen, 2015-09-24:
LCM is set to be a CMake pod, and has its source dir set to
```
${PROJECT_SOURCE_DIR}/externals/lcm/lcm-1.0.0
```

The clean-${proj} targets are set up to essentially call make clean in their project's source dir. But the lcm/lcm-1.0.0 directory doesn't contain a Makefile; its parent directory does.

What's the right way to fix this? Related: how do we define a cmake pod? I feel like it should be a pod that has a CMakeLists.txt in the root directory, not in some other place.

RussTedrake, 2015-09-24:
I think it's time to male the cmake branch of lcm to master.  We can move everything from  lcm-1.0.0 up into the root dir at the same time.

jhoare, 2016-03-03:
I'm looking at this right now. The lcm-pod has a Makefile in the root that is set to build lcm with cmake. I'm able to build the system (and then make clean-lcm it)  when I change it to be Not a CMake project from the superbuild. Is this the right way to do this, I'm trying to make sure this fits into the "pods" way. 

As an aside, it looks like a lot of the pods fail when doing clean-${proj}, such as bullet, and yaml_cpp

tkoolen, 2016-03-04:
Awesome, thank you for taking a look at this! I think not having it be a CMake pod is fine.

This, and the other failures when doing clean-${proj} are currently also causing `make clean` in drake-distro to fail, which is extremely annoying.

Related: https://github.com/RobotLocomotion/drake/issues/1338

PR #1819 (2016-03-08): Lcm make clean and appveyor update
Fixes #1368

Updates appveyor to install pkg-config and make from chocolatey.
closed 2016-03-08 by RussTedrake