RobotLocomotion/drake #1376: terrainInterpTest shows obvious mismatch with the terrain as visualized in the director

RussTedrake, 2015-09-26:
The dynamics are self-consistent and pass all of the matlab-side tests, but the visualization is obviously wrong.  The most likely culprit, I think, it my obj writing code.  @patmarion -- it's low priority, but tagging you in case you have ideas.

The test is drake/systems/plants/test/terrainInterpTest.m 
[image]

RussTedrake, 2015-09-27:
could be related to #1296 ?

patmarion, 2015-09-28:
I ran it on linux and the terrain mesh looks correct.  I'll try mac.

patmarion, 2015-09-28:
Looks correct on Mac, too.  So unfortunately I'm not able to reproduce what you've posted in the attached screenshot :(  For me, the .obj file seems to be written correctly on both platforms and displayed correctly.  It is:

v -7.000000 -3.000000 0.000000
v 7.000000 -3.000000 0.000000
v -7.000000 3.000000 0.000000
v 7.000000 3.000000 5.000000
vt 0.000000 0.000000
vt 1.000000 0.000000
vt 0.000000 1.000000
vt 1.000000 1.000000
g mesh
f 1/1 2/2 4/4 3/3
g

RussTedrake, 2015-09-28:
Thanks for checking Pat.  I’ve only done it on my mac.  

Does it always look ok if you run it a few times?  (the intial conditions are random, and only some land in the “air”)

patmarion, 2015-09-28:
Yes, I tried running it several times.  From what I can tell in the source code, the terrain geometry is not randomized, only the initial state of the brick.  The coordinates written in the obj file appear to match those defined in the test file:

options.terrain = RigidBodyHeightMapTerrain([-7,7],[-3,3],5*[0 0; 0 1]);

but in your screenshot, the terrain geometry seems to be wrong, so I'm not sure what that could be.  Given that Mike reported #1296, maybe some machines are misreading the obj files as written, and others are not?

RussTedrake, 2016-05-01:
this is still clearly wrong on my mac.

patmarion, 2016-05-01:
can you check the .obj file that is written to disk, does it contain the same values I posted in https://github.com/RobotLocomotion/drake/issues/1376#issuecomment-143782730 ?

RussTedrake, 2016-05-01:
i see 
```
v -7.000000 -3.000000 0.000000
v 7.000000 -3.000000 0.000000
v -7.000000 3.000000 0.000000
v 7.000000 3.000000 5.000000
vt 0.000000 0.000000
vt 1.000000 0.000000
vt 0.000000 1.000000
vt 1.000000 1.000000
g mesh
f 1/1 2/2 4/4 3/3
g
```

patmarion, 2016-05-01:
Ah, I see what's going on.  The obj file specifies a quadrilateral:

    f 1/1 2/2 4/4 3/3

But the quad vertices are not coplanar.  When the quad is rendered it must be split into triangles, and depending on how it is split, you will get one of the following two shapes:

[image]

[image]

Maybe the obj writer should explicitly write the triangles faces.  These are the two possible triangulations:

    f 1/1 2/2 4/4
    f 1/1 4/4 3/3

or

    f 1/1 3/3 2/2
    f 3/3 4/4 2/2

If you edit your .obj file to use one of these face definitions, you can load it into drake-visualizer by using the "File --> Open Data..." menu to test.

RussTedrake, 2016-05-01:
I’ve made the change to the matlab obj writer and it does fix this example.  Will send a PR momentarily.

PR #2230 (2016-05-10): Minor fixes for matlab <-> drake-visualizer interactions
resolves #1376 and #1452 
also cleans up the writeMeshOBJ, which was computing normals then never using them.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2230)
commit 6edb178a (2016-05-10): resolves #1376
closed 2016-05-10 by sherm1