RobotLocomotion/drake #1378: PR2 test prints "No closest point found" for every collisionDetect call after compile and the first dt

RussTedrake, 2015-09-26:
very suspicious.  perhaps something is not quite right with the new kinematics cache + bullet?

examples/PR2/runPassive

avalenzu, 2015-09-29:
Is this new behavior?

RussTedrake, 2015-09-29:
Yes.  Somehow it’s much worse with the quaternion floating base coordinates…

saturday (before merge):
http://kobol.csail.mit.edu/cd/testDetails.php?test=2896251&build=16389 <http://kobol.csail.mit.edu/cd/testDetails.php?test=2896251&build=16389>

saturday (on the pull request):
http://kobol.csail.mit.edu/cd/testDetails.php?test=2892466&build=16375 <http://kobol.csail.mit.edu/cd/testDetails.php?test=2892466&build=16375>

tkoolen, 2015-10-15:
BulletModel::findClosestPointsBtwElements
(just so people can get here by searching for it.)

tkoolen, 2015-10-15:
I think this is because we call
```
      [phi,~,~,~,idxA,idxB] = model.collisionDetect(zeros(model.getNumPositions,1));
```
on line 850 of RigidBodyManipulator.m. For quat floating joints, all zeros is not a valid configuration, and you can't even fix it by normalizing the quaternion in this case.

Should we introduce a method named e.g. nominalConfiguration or zeroConfiguration? Preferences as to the name?

RussTedrake, 2015-10-15:
good call.  that might be it.  
for what it's worth, we do have `getRandomConfiguration()` , which supports quats.  the names should be consistent so I would say `getZeroConfiguration()`.  or decide to change the other method to `randomConfiguration()`.

RussTedrake, 2015-12-29:
I just realized that this is actually the reason that drake006 has become so much less effective as a build server.  It's (slowly) writing comically enormous log files filled with reams of this error message.  I've marked this as high priority.

closed 2016-01-08 by RussTedrake