RobotLocomotion/drake #1379: Windows Binary Installation DCSFunction.mexw64 module error

dsukhin, 2015-09-27:
After downloading the Windows binary version of Drake, unzipping the folder and running install.sh, and running addpath_drake in Matlab (as provided in these instructions: https://github.com/RobotLocomotion/drake/wiki/FromBinary), I am still unable to run any of the examples. 

They fail with the following error:

<pre><code>>> cd PlanarMonopodHopper\
>> StateMachineControl.run()
Error using DrakeSystem/getModel (line 226)
Error while obtaining sizes from MEX S-function 'DCSFunction' in 'HopperPlant_0954912/DrakeSys'.

Error in DynamicalSystem/feedback (line 151)
      mdl.addSubsystem('system1',sys1.getModel());

Error in DrakeSystem/feedback (line 542)
            sys = feedback@DynamicalSystem(sys1,sys2);

Error in StateMachineControl.run (line 107)
      y=simulate(feedback(h,c),[0 3]);

Caused by:
    Error using DrakeSystem/getModel (line 226)
    Invalid MEX-file 'C:\Users\...\drake-0.9.9-6f72c04ab13168144148c64b9f90f4d4a4c28d36-win64\drake\systems\DCSFunction.mexw64':
    The specified module could not be found.
</code></pre>
 
I can verify that the file with the issue (DCSFunction.mexw64) is present at the given location. I am running Matlab R2015b on Windows 7 x64.

Thank you for any help!

RussTedrake, 2015-09-27:
if you are willing, it would be super helpful to run dependency walking on that file, from inside matlab.  I downloaded it from http://dependencywalker.com/ and then ran it (again, from inside matlab, so that the path, etc is correct) like this:

```
>> !z:\russt\depends.exe ..\..\systems\DCSFunction.mexw64
```

My hopper runs fine… even though dependency walker still complains about a few things.  But this is the best tool I can think of for us to help remotely.

dsukhin, 2015-09-27:
Sure thing. Here is a screenshot of the walk. It looks like MSVCR120.DLL is its primary complaint. Let me know if you would like me to dive deeper into any section. Thank you!
[image]

RussTedrake, 2015-09-27:
great.  that's probably it.  can you install it from here:
http://download.microsoft.com/download/9/1/4/914851c6-9141-443b-bdb4-8bad3a57bea9/vcredist_x64.exe

(and make sure it makes it into your path... you probably have to restart matlab)

dsukhin, 2015-09-27:
Hmm... Should I check that it's in the Windows path, Matlab path, or otherwise? What is the best way to verify or add it to the correct place? I ran the installer above and it finished without issue but it seems to not have changed anything... the depends still gives the same complaints and the examples still do not run.

RussTedrake, 2015-09-27:
the windows path (then restart matlab) is safe.

dsukhin, 2015-09-28:
Ok. It looks like the link you provided installed another variant of MSVCR (not 120). http://www.microsoft.com/en-us/download/details.aspx?id=40784 however put the right version into windows32 (already in the path). Just restarted Matlab and things run now. Thank you for your help.

RussTedrake, 2015-09-28:
great.  thanks.  will reopen and assign so that we make sure to package that with our distro.

RussTedrake, 2015-10-08:
@psiorx  -- is this fix also included in the latest windows binaries?

psiorx, 2015-10-08:
Yes.  Director also required msvcr120.dll so it's all bundled now.

closed 2015-10-08 by psiorx