RobotLocomotion/drake #1381: Windows Binary: Runnnig install.sh in a path with spaces

dsukhin, 2015-09-28:
When running install.sh (to sed all the absolute paths to the new location), the sed command fails if there is a space in the new path. The files that were found are then replaced with empty files when the following mv command ran. This variable whitespace in the new path needs to be escaped before running sed.

<pre><code>sed: -e expression #1, char 36: unknown option to `s'</code></pre>

On a related note, when I modified the paths by hand, the eigen3.pc files do not have the the paths quoted so they will also fail if replaced with a path with a space.

RussTedrake, 2015-10-04:
@rdeits , @psiorx  -- didn't you fix this one?

psiorx, 2015-10-05:
We fixed one of the issues but not all of them.

I'm working on a solution that should make the install script unnecessary by making the CMake generated MATLAB functions smarter.  Hopefully I can release those in a 0.9.10 build today.

RussTedrake, 2015-10-08:
@psiorx -- the install script is gone now, right?

psiorx, 2015-10-08:
Yes.

closed 2015-10-08 by psiorx