RobotLocomotion/drake #1382: drake-distro make: 'drake/lcmt_zmp_data.hpp' file not found #include "drake/lcmt_zmp_data.hpp"

tarquasso, 2015-09-28:
when running make in drake-distro after having updated the complete drake distro to current, getting this error:

[ 33%] Building CXX object util/CMakeFiles/drakeLCMUtil.dir/lcmUtil.cpp.o
In file included from /Users/rkk/Documents/MATLAB/drake-distro/drake/drake/util/lcmUtil.cpp:1:
In file included from /Users/rkk/Documents/MATLAB/drake-distro/drake/drake/util/./lcmUtil.h:10:
/Users/rkk/Documents/MATLAB/drake-distro/build/include/lcmtypes/drake/lcmt_qp_controller_input.hpp:14:10: fatal error: 'drake/lcmt_zmp_data.hpp' file not found
#include "drake/lcmt_zmp_data.hpp"
         ^
1 error generated.
make[7]: *** [util/CMakeFiles/drakeLCMUtil.dir/lcmUtil.cpp.o] Error 1
make[6]: *** [util/CMakeFiles/drakeLCMUtil.dir/all] Error 2
make[5]: *** [all] Error 2
make[4]: *** [drake-prefix/src/drake-stamp/drake-build] Error 2
make[3]: *** [CMakeFiles/drake.dir/all] Error 2
make[2]: *** [all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2

But I can find the file here:
drake-distro/build/include/lcmtypes/drake

and here:

drake-distro/drake/drake/pod-build/lcmgen/lcmtypes/drake

on my computer.

Shall I add includes somewhere, or what else can I do to fix?

rdeits, 2015-09-28:
Did you update from a fairly old (more than a couple of months) version of Drake? It looks like you might have some stale lcmtype headers lying around from before https://github.com/RobotLocomotion/cmake/pull/17 

Can you try deleting `drake/pod-build/lcmgen/` and `build/include/lcmtypes` and running `make` again?

tarquasso, 2015-09-28:
> Did you update from a fairly old (more than a couple of months) version of Drake? It looks like you might have some stale lcmtype headers lying around from before RobotLocomotion/cmake#17

yes, I got help by Twan to update, these steps:

git checkout your-branch

# rebase your branch onto the last commit before the great move
git rebase d3ff393f9639135c6d421a311e789773fdc74701

# fix any conflicts

# create a format patch containing all of your commits
git format-patch d3ff393f9639135c6d421a311e789773fdc74701 --stdout > ~/patch.patch

# create a branch that will contain your changes, moved, currently pointing to RobotLocomotion master just after the great move:
git checkout -b your-branch-moved f47c72e007588c0545b8c03212809ca7b847a7f1

# apply patches in the new drake subdirectory of the drake repository:
git am --ignore-whitespace --directory drake ~/patch.patch

# rebase onto (current) master (assuming master points to RobotLocomotion master)
git rebase master

I will delete those folders and then run make again.

Cheers
Robert

tarquasso, 2015-09-28:
ok, deleted it, make now runs further, now got to this error:

[ 52%] Building CXX object systems/plants/collision/CMakeFiles/drakeCollision.dir/BulletModel.cpp.o
/Users/rkk/Documents/MATLAB/drake-distro/drake/drake/systems/plants/collision/BulletModel.cpp:124:99: error: too many arguments to function call, expected single argument 'point', have 2
      arguments
        bt_convex_hull_shape->addPoint(btVector3(vertices(0, i), vertices(1, i), vertices(2, i)), false);
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                            ^~~~~
/Users/rkk/Documents/MATLAB/drake-distro/build/include/bullet/BulletCollision/CollisionShapes/btConvexHullShape.h:39:2: note: 'addPoint' declared here
        void addPoint(const btVector3& point);
        ^
/Users/rkk/Documents/MATLAB/drake-distro/drake/drake/systems/plants/collision/BulletModel.cpp:143:118: error: too many arguments to function call, expected single argument 'point', have 2
      arguments
      bt_convex_hull_shape->addPoint(btVector3(geometry.points(0, i), geometry.points(1, i), geometry.points(2, i)), false);
      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                                 ^~~~~
/Users/rkk/Documents/MATLAB/drake-distro/build/include/bullet/BulletCollision/CollisionShapes/btConvexHullShape.h:39:2: note: 'addPoint' declared here
        void addPoint(const btVector3& point);
        ^
2 errors generated.
make[7]: *** [systems/plants/collision/CMakeFiles/drakeCollision.dir/BulletModel.cpp.o] Error 1
make[6]: *** [systems/plants/collision/CMakeFiles/drakeCollision.dir/all] Error 2
make[5]: *** [all] Error 2
make[4]: *** [drake-prefix/src/drake-stamp/drake-build] Error 2
make[3]: *** [CMakeFiles/drake.dir/all] Error 2
make[2]: *** [all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2

RussTedrake, 2015-09-28:
you probably have an old version of bullet?

tarquasso, 2015-09-28:
yes, it was old. running make again... will have to wait a while because I ran make clean earlier.

Is there a better way to check for updates of the various subdirectory repositories than running this command from the drake-distro folder?:

find . -type d -depth 1 -exec git --git-dir={}/.git --work-tree=$PWD/{} pull origin master \;

rdeits, 2015-09-28:
Running `make` from drake-distro should update and build all your
subdirectories, and running `make download-all` should download the correct
version of each one without compiling it.

tarquasso, 2015-09-28:
new error: 

tarqu:mosek rkk$ git pull origin master
From https://github.com/RobotLocomotion/mosek
 * branch            master     -> FETCH_HEAD
Already up-to-date.
tarqu:mosek rkk$ make

BUILD_PREFIX: /Users/rkk/Documents/MATLAB/drake-distro/build

CMake Error at CMakeLists.txt:7 (include):
  include could not find load file:

    cmake/pods.cmake

CMake Error at CMakeLists.txt:8 (include):
  include could not find load file:

    cmake/matlab_pods.cmake

CMake Error at CMakeLists.txt:52 (pods_install_headers):
  Unknown CMake command "pods_install_headers".

-- Configuring incomplete, errors occurred!
See also "/Users/rkk/Documents/MATLAB/drake-distro/mosek/pod-build/CMakeFiles/CMakeOutput.log".
make[1]: *** [configure] Error 1
make: *** [pod-build/Makefile] Error 2

These files are actually not in the cmake folder of mosek:
    cmake/pods.cmake
cmake/matlab_pods.cmake

RussTedrake, 2015-09-28:
you are building each directory manually, and don’t have the updated versions.  run make from the drake-distro directory.

tarquasso, 2015-09-28:
The error I copied above was generated after running make from the drake-distro folder

I think there are several problems in the repos within my drake-distro folder, for example if I run:
$ make download-all

I get this:

-------------------------------------------
-- lcm
-------------------------------------------
make[1]: *** No rule to make target `download-all'.  Stop.

-------------------------------------------
-- libbot
-------------------------------------------
make[1]: *** No rule to make target `download-all'.  Stop.

-------------------------------------------
-- eigen
-------------------------------------------
make[2]: *** No rule to make target `download-all'.  Stop.
make[1]: *** [download-all] Error 2

-------------------------------------------
-- bullet
-------------------------------------------
make[2]: *** No rule to make target `download-all'.  Stop.
make[1]: *** [download-all] Error 2

-------------------------------------------
-- octomap
-------------------------------------------
make[2]: *** No rule to make target `download-all'.  Stop.
make[1]: *** [download-all] Error 2

-------------------------------------------
-- snopt
-------------------------------------------
make[2]: *** No rule to make target `download-all'.  Stop.
make[1]: *** [download-all] Error 2

-------------------------------------------
-- gurobi
-------------------------------------------
make[2]: *** No rule to make target `download-all'.  Stop.
make[1]: *** [download-all] Error 2

-------------------------------------------
-- mosek
-------------------------------------------
make[2]: *** No rule to make target `download-all'.  Stop.
make[1]: *** [download-all] Error 2

-------------------------------------------
-- yalmip
-------------------------------------------
make[1]: *** No rule to make target `download-all'.  Stop.

-------------------------------------------
-- gloptipoly3
-------------------------------------------
make[1]: *** No rule to make target `download-all'.  Stop.

-------------------------------------------
-- bertini
-------------------------------------------
make[1]: *** No rule to make target `download-all'.  Stop.

-------------------------------------------
-- sedumi
[... 71 lines cut ...]
[ 51%] Completed 'download-libbot'
[ 58%] Built target download-libbot
[ 60%] Performing update step for 'download-bullet'
[ 62%] No configure step for 'download-bullet'
[ 65%] No build step for 'download-bullet'
[ 67%] No install step for 'download-bullet'
[ 67%] Completed 'download-bullet'
[ 72%] Built target download-bullet
[ 74%] Performing update step for 'download-cmake'
[ 74%] No configure step for 'download-cmake'
[ 76%] No build step for 'download-cmake'
[ 76%] No install step for 'download-cmake'
[ 79%] Completed 'download-cmake'
[ 86%] Built target download-cmake
[ 88%] Performing update step for 'download-lcm'
[ 90%] No configure step for 'download-lcm'
[ 93%] No build step for 'download-lcm'
[ 95%] No install step for 'download-lcm'
[ 95%] Completed 'download-lcm'
[100%] Built target download-lcm
[100%] Built target download-all

For example, when comparing the contents of the mosek folder online with what I have locally, they do not match up. the mosek/cmake folder is empty, whereas there should be a bunch of fuiles. running git pull origin master does not help this either. The patching has not done any good to my drake install...

I am now considering just checking out all of drake-distro from scratch and then patch my drake subfolder with my specific changes. Or any other suggestions before I pull the brake?

tarquasso, 2015-09-28:
bye the way, this is shown when I am running git status in drake-distro:

k$ git status
On branch master
Your branch is up-to-date with 'origin/master'.
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)

	modified:   bertini (new commits)
	modified:   bullet (new commits)
	modified:   director (new commits)
	modified:   drake (new commits)
	modified:   eigen (new commits)
	modified:   gloptipoly3 (new commits)
	modified:   gurobi (new commits, modified content)
	modified:   iris (new commits)
	modified:   lcm (new commits)
	modified:   libbot (new commits)
	modified:   mosek (new commits)
	modified:   sedumi (new commits)
	modified:   snopt (new commits)
	modified:   spotless (new commits, modified content)
	modified:   yalmip (new commits)

no changes added to commit (use "git add" and/or "git commit -a")

and I don't know how to do anything about it.

RussTedrake, 2015-09-28:
> (use "git add ..." to update what will be committed)
> (use "git checkout -- ..." to discard changes in working directory)

you should zap all of those directories (except drake! :)  

since the switch to superbuild, your root drake-distro folder should look like

```
drake008% ls -l
total 80
-rw-r--r--   1 russt  staff  16576 Sep 28 16:23 CMakeLists.txt
-rw-r--r--   1 russt  staff   1583 Sep  6 05:06 LICENSE.TXT
-rw-r--r--   1 russt  staff   2208 Sep 23 20:41 Makefile
-rw-r--r--   1 russt  staff    498 Sep 28 16:29 README.md
-rw-r--r--   1 russt  staff    399 Aug  2 19:37 addpath_pods.m
drwxr-xr-x   8 russt  staff    272 Sep 27 11:25 build
drwxr-xr-x  22 russt  staff    748 Sep 28 16:31 drake
drwxr-xr-x  22 russt  staff    748 Sep 26 12:30 externals
-rwxr-xr-x   1 russt  staff    681 Aug 22 13:41 install_prereqs.sh
drwxr-xr-x  50 russt  staff   1700 Sep 27 12:17 pod-build
drake008% pwd
/Users/russt/drake-distro
```

closed 2015-09-29 by RussTedrake