RobotLocomotion/drake #1386: ddapp tries to configure (and fails) if bot-lcmgl-client is not present

tarquasso, 2015-09-30:
running
`cd drake-distro`  
`make`
with `libbot` option deactivated by running `make options` leads to:

```[ 90%] Performing configure step for 'ddapp'
loading initial cache file /Users/rkk/Documents/MATLAB/drake-distro/externals/director/distro/pods/drake-distro/pod-build/tmp/ddapp-cache-Release.cmake
-- The C compiler identification is AppleClang 6.1.0.6020053
-- The CXX compiler identification is AppleClang 6.1.0.6020053
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenGL: /System/Library/Frameworks/OpenGL.framework  
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "0.28") 
-- checking for module 'bot2-lcmgl-client'
--   package 'bot2-lcmgl-client' not found
CMake Error at cmake/dd-macros.cmake:36 (message):
  required package bot2-lcmgl-client not found.
  PKG_CONFIG_PATH=/Users/rkk/Documents/MATLAB/drake-distro/build/lib/pkgconfig::/opt/X11/lib/pkgconfig
Call Stack (most recent call first):
  src/vtk/DRCFilters/CMakeLists.txt:140 (use_pkg)

-- checking for module 'bot2-lcmgl-renderer'
--   package 'bot2-lcmgl-renderer' not found
CMake Error at cmake/dd-macros.cmake:36 (message):
  required package bot2-lcmgl-renderer not found.
  PKG_CONFIG_PATH=/Users/rkk/Documents/MATLAB/drake-distro/build/lib/pkgconfig::/opt/X11/lib/pkgconfig
Call Stack (most recent call first):
  src/vtk/DRCFilters/CMakeLists.txt:140 (use_pkg)

CMake Warning (dev) at /usr/local/opt/vtk5/lib/vtk-5.10/CMake/vtkWrapPython.cmake:147 (CONFIGURE_FILE):
  configure_file called with unknown argument(s):

   COPY_ONLY

Call Stack (most recent call first):
  cmake/wrap-python.cmake:12 (vtk_wrap_python3)
  src/vtk/DRCFilters/CMakeLists.txt:151 (wrap_python)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Looking for Q_WS_X11
-- Looking for Q_WS_X11 - not found
-- Looking for Q_WS_WIN
-- Looking for Q_WS_WIN - not found
-- Looking for Q_WS_QWS
-- Looking for Q_WS_QWS - not found
-- Looking for Q_WS_MAC
-- Looking for Q_WS_MAC - found
-- Looking for QT_MAC_USE_COCOA
-- Looking for QT_MAC_USE_COCOA - found
-- Found Qt4: /usr/local/bin/qmake (found version "4.8.7") 
-- Found PythonLibs: /System/Library/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib (found version "2.7.6") 
-- Found PythonInterp: /usr/bin/python2.7 (found version "2.7.6") 
-- Found PythonQt: /Users/rkk/Documents/MATLAB/drake-distro/build/include/PythonQt  
-- Found CTKPythonConsole: /Users/rkk/Documents/MATLAB/drake-distro/build/include/ctkPythonConsole  
-- Found QtPropertyBrowser: /Users/rkk/Documents/MATLAB/drake-distro/build/include/QtPropertyBrowser  
-- checking for module 'lcm'
--   found lcm, version 1.0.0
-- Found LCM: /Users/rkk/Documents/MATLAB/drake-distro/build/include  
-- Found NumPy: version "1.8.0rc1" /System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/numpy/core/include
-- Found Doxygen: /usr/local/bin/doxygen (found version "1.8.10") 
-- sphinx-build not found - python API documentation will not be built.
-- Configuring incomplete, errors occurred!
See also "/Users/rkk/Documents/MATLAB/drake-distro/externals/director/distro/pods/drake-distro/pod-build/src/ddapp-build/CMakeFiles/CMakeOutput.log".
See also "/Users/rkk/Documents/MATLAB/drake-distro/externals/director/distro/pods/drake-distro/pod-build/src/ddapp-build/CMakeFiles/CMakeError.log".
make[7]: *** [src/ddapp-stamp/ddapp-configure] Error 1
make[6]: *** [CMakeFiles/ddapp.dir/all] Error 2
make[5]: *** [all] Error 2
make[4]: *** [all] Error 2
make[3]: *** [director-prefix/src/director-stamp/director-build] Error 2
make[2]: *** [CMakeFiles/director.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2```

what to do?

patmarion, 2015-09-30:
The director depends on libbot.  If you have disabled libbot, you must also disable director.  I'm not sure if there is a method in drake's cmake code for detecting and avoiding these dependency issues.

patmarion, 2015-10-02:
@RussTedrake I think you could close this issue.  I opened a new issue for the general problem here: https://github.com/RobotLocomotion/drake/issues/1390

closed 2015-10-02 by RussTedrake