RobotLocomotion/drake #1387: Drake Ubuntu Installation Instructions

vtjeng, 2015-09-30:
I attempted to install `drake` on a new installation of Ubuntu 14.04.

Following the instructions on this page: https://github.com/RobotLocomotion/drake/wiki/Ubuntu, I ran the following commands:

    sudo apt-get update
    sudo apt-get install cmake
    cd drake-distro

However, `make options` failed, with the error message `CMAKE_CXX_COMPILER-notfound`.

Working based on the solution in here: http://askubuntu.com/questions/152653/what-should-i-do-about-this-error, I ran the command `sudo apt-get install build-essential`, but `make options` still failed, with the following error message:

    Configuring with CMAKE_FLAGS: -DCMAKE_BUILD_TYPE=Release
    -- The CXX compiler identification is GNU 4.8.4
    -- Check for working CXX compiler: /usr/bin/c++
    -- Check for working CXX compiler: /usr/bin/c++ -- works
    -- Detecting CXX compiler ABI info
    -- Detecting CXX compiler ABI info - done
    -- CMAKE_INSTALL_PREFIX=/home/vtjeng/drake-distro/build
    -- Preparing to build cmake with dependencies: 
    -- Preparing to build eigen with dependencies: 
    -- Preparing to build lcm with dependencies: 
    -- Preparing to build libbot with dependencies: lcm
    -- Preparing to build bullet with dependencies: 
    -- Preparing to build director with dependencies: lcm;libbot
    -- Preparing to build drake with dependencies: cmake;eigen;lcm;libbot;bullet
    -- Configuring done
    -- Generating done
    -- Build files have been written to: /home/vtjeng/drake-distro/pod-build
    make: ccmake: Command not found
    make: *** [options] Error 127

`make options` finally worked after I ran `sudo apt-get install cmake-curses-gui`.

## Summary
We might want to add the following lines to the installation instructions:

    sudo apt-get install build-essential
    sudo apt-get install cmake-curses-gui

rdeits, 2015-09-30:
Good point. We've got something of a dependency loop because you need build-essential and ccmake to do `make options` but you need to do `make options` in order to get all of your dependencies. 

Perhaps we can have people do:

```
./install_dependencies.sh 
make options
make download-all
./install_dependencies.sh
```

?

vtjeng, 2015-09-30:
I don't have enough experience to know whether that would work.

As a side note, the correct command to run (as of today) seems to be 

    sudo ./install_prereqs.sh ubuntu

rather than `sudo ./install_prerequisites.sh ubuntu` as in the wiki.

People should probably be able to figure this out for themselves, but if we're editing the instructions, we might as well change this.

psiorx, 2015-09-30:
Thanks for the info.  I've updated the wiki page to reflect your findings.

closed 2015-10-04 by RussTedrake