RobotLocomotion/drake #1389: Windows 7 / Matlab 2015b / MSVC 2013: Matlab crash constructing Acrobot from URDF

tkoolen, 2015-10-01:
From Piazza, but I can reproduce on my machine.

For the combination Windows 7 / Matlab 2015b / MSVC 2013 / drake master:
```
p = PlanarRigidBodyManipulator('Acrobot.urdf');
```
results in a Matlab crash. Replacing `PlanarRigidBodyManipulator` with `RigidBodyManipulator` also results in a crash.

The crash happens at (or just after) the constructModelmex call. But it may just be the first mex call.

Crash file:
```
MATLAB crash file:C:\Users\Twan\AppData\Local\Temp\matlab_crash_dump.5676-1:

------------------------------------------------------------------------
          Access violation detected at Thu Oct 01 18:23:04 2015
------------------------------------------------------------------------

Configuration:
  Crash Decoding      : Disabled
  Crash Mode          : continue (default)
  Current Graphics Driver: Unknown hardware 
  Default Encoding    : windows-1252
  Graphics card 1     : NVIDIA ( 0x10de ) NVIDIA GeForce GT 650M Version 9.18.13.3228
  Host Name           : Sloth
  MATLAB Architecture : win64
  MATLAB Root         : C:\Program Files\MATLAB\R2015b
  MATLAB Version      : 8.6.0.267246 (R2015b)
  OpenGL              : hardware
  Operating System    : Microsoft Windows 7 Professional 
  Processor ID        : x86 Family 6 Model 58 Stepping 9, GenuineIntel
  Virtual Machine     : Java 1.7.0_60-b19 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
  Window System       : Version 6.1 (Build 7601: Service Pack 1)

Fault Count: 1

Abnormal termination:
Access violation

Register State (from fault):
  RAX = 000000fffffffffe  RBX = 00000cafa3ae7730
  RCX = 000000ffffffffff  RDX = 0000000000000000
  RSP = 0000000004028f70  RBP = 00000000abeb79c0
  RSI = 0000000000000000  RDI = 0000000000000000
 
   R8 = 00000000abeb79b0   R9 = 000000006e88b0e0
  R10 = 00000000ab5d173a  R11 = 00000000abeb6cac
  R12 = 00000000041a0000  R13 = 00000000ac037c9c
  R14 = 000000006e88b0e0  R15 = 0000000000000000
 
  RIP = 0000000076d50e91  EFL = 00010206
 
   CS = 0033   FS = 0053   GS = 002b

Stack Trace (from fault):
[  0] 0x0000000076d50e91                      C:\Windows\SYSTEM32\ntdll.dll+00331409 RtlAnsiStringToUnicodeString+00000305
[  1] 0x0000000076b01a7a                   C:\Windows\system32\kernel32.dll+00137850 HeapFree+00000010
[  2] 0x000007fefaf86338 C:\Program Files\MATLAB\R2015b\bin\win64\MSVCR110.dll+00418616 free+00000028
[  3] 0x000000006e7530b0 C:\Program Files\MATLAB\R2015b\bin\win64\icuuc54.dll+00667824 icu_54::UnicodeString::~UnicodeString+00000048
[  4] 0x000000006e811748 C:\Program Files\MATLAB\R2015b\bin\win64\icuin54.dll+00005960 icu_54::UStringSet::~UStringSet+00000584
[  5] 0x000000006e88b739 C:\Program Files\MATLAB\R2015b\bin\win64\icuin54.dll+00505657 icu_54::Formattable::dispose+00000057
[  6] 0x000000006e88b105 C:\Program Files\MATLAB\R2015b\bin\win64\icuin54.dll+00504069 icu_54::Formattable::~Formattable+00000037
[  7] 0x00000000fff8a991 C:\Program Files\MATLAB\R2015b\bin\win64\libmwi18n.dll+00567697 fl::i18n::detail::get_module_path_from_address+00002449
[  8] 0x00000000fff1402e C:\Program Files\MATLAB\R2015b\bin\win64\libmwi18n.dll+00081966 fl::filesystem::file_status::operator=+00002862
[  9] 0x00000000fff19d7a C:\Program Files\MATLAB\R2015b\bin\win64\libmwi18n.dll+00105850 fl::filesystem::file_status::exists+00002106
[ 10] 0x00000000fff1a466 C:\Program Files\MATLAB\R2015b\bin\win64\libmwi18n.dll+00107622 fl::i18n::MessageCatalog::get_message+00000214
[ 11] 0x000000000eb58a99 C:\Program Files\MATLAB\R2015b\bin\win64\mcos_impl.dll+00821913 QueryMLFcnTable_mcos_impl+00399609
[ 12] 0x000000000eb59dcf C:\Program Files\MATLAB\R2015b\bin\win64\mcos_impl.dll+00826831 QueryMLFcnTable_mcos_impl+00404527
[ 13] 0x000000000eb5933a C:\Program Files\MATLAB\R2015b\bin\win64\mcos_impl.dll+00824122 QueryMLFcnTable_mcos_impl+00401818
[ 14] 0x000000000eaafbf6 C:\Program Files\MATLAB\R2015b\bin\win64\mcos_impl.dll+00130038
[... 47 lines cut ...]
[ 62] 0x00000000044e0ad3 C:\Program Files\MATLAB\R2015b\bin\win64\libmwservices.dll+01510099 services::system_events::PpeDispatchHook::dispatchOne+00032291
[ 63] 0x00000000044ec785 C:\Program Files\MATLAB\R2015b\bin\win64\libmwservices.dll+01558405 sysq::addProcessPendingEventsUnitTestHook+00006101
[ 64] 0x00000000044ec830 C:\Program Files\MATLAB\R2015b\bin\win64\libmwservices.dll+01558576 sysq::addProcessPendingEventsUnitTestHook+00006272
[ 65] 0x00000000044ee475 C:\Program Files\MATLAB\R2015b\bin\win64\libmwservices.dll+01565813 sysq::getCondition+00004197
[ 66] 0x00000000044efe22 C:\Program Files\MATLAB\R2015b\bin\win64\libmwservices.dll+01572386 svWS_ProcessPendingEvents+00000162
[ 67] 0x000000000864d3cd   C:\Program Files\MATLAB\R2015b\bin\win64\mcr.dll+00185293 mcr::runtime::setInterpreterThreadToCurrent+00023677
[ 68] 0x000000000864e03a   C:\Program Files\MATLAB\R2015b\bin\win64\mcr.dll+00188474 mcr::runtime::setInterpreterThreadToCurrent+00026858
[ 69] 0x0000000008645c95   C:\Program Files\MATLAB\R2015b\bin\win64\mcr.dll+00154773 mcr_process_events+00007557
[ 70] 0x00000000086475a2   C:\Program Files\MATLAB\R2015b\bin\win64\mcr.dll+00161186 mcr_process_events+00013970
[ 71] 0x0000000006fb37ae C:\Program Files\MATLAB\R2015b\bin\win64\MVMLocal.dll+00210862 mvm_server::inproc::LocalFactory::terminate+00081678
[ 72] 0x00000000fa924d99   C:\Program Files\MATLAB\R2015b\bin\win64\mvm.dll+01002905 mvm::detail::initLocalMvmHack+00000537
[ 73] 0x00000000fa925368   C:\Program Files\MATLAB\R2015b\bin\win64\mvm.dll+01004392 mvm::detail::SessionImpl::privateSession+00000376
[ 74] 0x00000000fa9255b2   C:\Program Files\MATLAB\R2015b\bin\win64\mvm.dll+01004978 mvm::detail::SessionImpl::privateSession+00000962
[ 75] 0x0000000140006247 C:\Program Files\MATLAB\R2015b\bin\win64\MATLAB.exe+00025159
[ 76] 0x0000000140007cf4 C:\Program Files\MATLAB\R2015b\bin\win64\MATLAB.exe+00031988
[ 77] 0x0000000076af5a4d                   C:\Windows\system32\kernel32.dll+00088653 BaseThreadInitThunk+00000013
[ 78] 0x0000000076d2b831                      C:\Windows\SYSTEM32\ntdll.dll+00178225 RtlUserThreadStart+00000033

If this problem is reproducible, please submit a Service Request via:
    http://www.mathworks.com/support/contact_us/

A technical support engineer might contact you with further information.

Thank you for your help.
```

RussTedrake, 2015-10-02:
wow.  am taking a look.

RussTedrake, 2015-10-02:
just tried compiling here in debug mode and got

..\RigidBodyManipulator.cpp : fatal error C1128: number of sections exceeded object file format limit: compile with /bigobj

I think we are going to need to address the size of RigidBodyManipulator.cpp (not sure if it’s the templates or just the massive number of methods we’ve put in the same .cpp file; perhaps both)

RussTedrake, 2015-10-02:
i can reproduce this now.  the error changes if compiled in debug mode (but it was actually very hard to compile in debug mode… msvc kept crashing but I ran it many times to get through :(  ).  

My current best guess is that we are blowing the stack allocation on windows with our RBM.dll ?  but I would be much happier if it was a memory error in our code which we could simply fix.  

The other obvious question is:  how on earth did this get past the build servers?

tkoolen, 2015-10-02:
The latest precompiled version seems to be fine at least, so that could be a starting point for a git bisect.

RussTedrake, 2015-10-04:
8 hours of trial and error memory debugging later... i've fixed it.  (the DrakeJoint classes were missing some EIGEN_MAKE_ALIGNED_OPERATOR_NEW statements).  pull request coming in a few.

tkoolen, 2015-10-05:
Woah, I didn't know that a missing EIGEN_MAKE_ALIGNED_OPERATOR_NEW could cause trouble at runtime; I thought it would just result in a compilation error. I had been working on that assumption at least.

RussTedrake, 2015-10-07:
pretty sure i've found this one and fixed it for real now.  short version: newer versions of matlab added the /Zp8 flag to the CXXFLAGS for MSVC.  ( https://msdn.microsoft.com/en-us/library/xh3e3fd0.aspx ).  That was messing with Eigen's attempts at controlling the alignment.  I'm now stripping that flag back off before compiling.  That explains why it worked ok when using the mex options from MATLAB 2012a, but not for newer MATLAB versions.

This flag was not present in MATLAB 2014a, but is present in MATLAB 2015a.  (i don't have a version of 2014b handy on windows)

Sigh...

tkoolen, 2015-10-07:
Wow. Good find. We should definitely report this to the Mathworks, because it's going to hurt a lot of people in other projects as well.

By the way, to clear something up regarding the EIGEN_ALIGNED_OPERATOR_NEW for DrakeJoint subclasses: it should only have to be in the base class according to forum discussions I've seen involving Eigen maintainers; the methods defined by the macro will be inherited by the derived classes.

RussTedrake, 2015-10-07:
that’s good to know.  i wasn’t sure, and saw no harm in putting it everywhere…

tkoolen, 2015-10-07:
There isn't any harm, but it's not necessary.

closed 2015-10-07 by RussTedrake