RobotLocomotion/drake #1391: Windows source build produces wrong answers for Acrobot linearization

rdeits, 2015-10-03:
This was reported by one of the students in 6.832:

```
q_0 = [pi;0]
q_dot_0 = [0;0]
x_0 = [q_0; q_dot_0]
u_0 = 0;
t = 0;
p = PlanarRigidBodyManipulator('Acrobot.urdf');
[A, B, C, D] = p.linearize(t, x_0, u_0);
mat2str(A)
```

Gives:
```
A = [0 0 1 0;0 0 0 1;12.3910107300968 -12.2683274555414 -3.72101461443614e-16 6.49789119237356e-16;-14.3171381406168 28.7446912283335 6.49789119237356e-16 -1.30041130144809e-15]
```

which is substantially different than the value of A returned by the non-Windows build. Apparently the precompiled Windows distro produces the correct value of A.

RussTedrake, 2015-10-04:
this must clearly be related to #1389 .  Something is going wrong with the memory on windows (in some place it crashes, on others it generates garbage).  working on it now.

RussTedrake, 2015-10-08:
resolved by #1396 . :)

closed 2015-10-08 by RussTedrake