RobotLocomotion/drake #1392: Using the Gurobi Optimizer

vtjeng, 2015-10-04:
During the installation process for Drake, if users choose to [install Gurobi from source](https://github.com/RobotLocomotion/drake/wiki/FromSource) via `make`, they are required to insert their Gurobi.com username. However, as an additional step to get Gurobi working, it seems necessary to get an academic license and then to store the Gurobi license key file associated with the license. If this is indeed the case, then it would be useful to ensure that users do this first.

## Summary
Indicate that Gurobi is enabled by default if you use the WITH_ALL_PUBLIC_EXTERNALS option, and add a section on adding the key, specifically:

    cd drake-distro
    ./externals/gurobi/gurobi604/linux64/bin/grbgetkey [key-number]

RussTedrake, 2015-12-27:
also discussed in #1523

jonathan-owens, 2016-05-14:
Closed. See https://github.com/RobotLocomotion/drake/pull/2286

closed 2016-05-14 by jonathan-owens