RobotLocomotion/drake #1400: the lcp solves fails if people don't have gurobi

RussTedrake, 2015-10-08:
solveLCPmex is not compiled without gurobi, and when folks are running e.g. fallingBrickLCP, fallingCapsuleTest, or fallingSnakeTest, they are getting:
```
Undefined function 'solveLCPmex' for input arguments of type 'DrakeMexPointer'.

Error in TimeSteppingRigidBodyManipulator/solveMexLCP (line 266)
        [z, Mqdn, wqdn, possible_contact_indices, possible_jointlimit_indices] = solveLCPmex(obj.manip.mex_model_ptr, kinsol.mex_ptr, u, phiC, n, D, obj.timestep, obj.z_inactive_guess_tol, obj.LCP_cache.data.z, H, C, B, obj.enable_fastqp);

Error in TimeSteppingRigidBodyManipulator/update (line 211)
        [obj2,z2,Mvn2,wvn2] = solveMexLCP(obj,t,x,u);

Error in SimulinkModelHandle/sim (line 112)
      varargout{:} = sim(obj.name,varargin{:});

Error in DynamicalSystem/simulate (line 120)
simout = sim(mdl,pstruct);

Error in DrakeSystem/simulate (line 494)
        [varargout{:}] = simulate@DynamicalSystem(obj,varargin{:});

Error in fallingBrickLCP (line 24)
xtraj = p.simulate([0 4],x0);
Error using SimulinkModelHandle/sim (line 112)
Error reported by S-function 'DCSFunction' in 'TimeSteppingRigidBodyManipulator_4427837/DrakeSys':

DrakeSystem S-Function: error MATLAB:UndefinedFunction in MATLAB callback.
See additional debugging information above

Error in DynamicalSystem/simulate (line 120)
simout = sim(mdl,pstruct);

Error in DrakeSystem/simulate (line 494)
        [varargout{:}] = simulate@DynamicalSystem(obj,varargin{:});

Error in fallingBrickLCP (line 24)
xtraj = p.simulate([0 4],x0);
```

in my view, there are two problems:  1) our check for solveLCPmex in TSRBM must not be sufficient, and 2) we should probably still have the non-gurobi lcp mex working even if gurobi is not present.

RussTedrake, 2015-10-08:
oops.  here are some examples from my snopt precompiled pull request:
http://drake006.csail.mit.edu/drake/pull-RussTedrake-snopt_precompiled_for_all-into-RobotLocomotion-master_20151008-1111-Continuous

psiorx, 2015-10-08:
> [obj2,z2,Mvn2,wvn2] = solveMexLCP(obj,t,x,u);

I think this has something to do with the code Michael added for supporting quaternion floating bases in contact simulations.  Looks like there's now a second call to solveMexLCP that circumvents all the checks that all the other one has.  I'll push a fix shortly.

PR #1403 (2015-10-08): made gurobi non-public and fixed bug in TSRBM
Hotfix for tsrbm.

Resolves #1400
closed 2015-10-08 by RussTedrake