RobotLocomotion/drake #1402: A few tests fail with missing dependencies even when the precompiled snopt is present

RussTedrake, 2015-10-08:
http://drake006.csail.mit.edu/drake/pull-RussTedrake-snopt_precompiled_for_all-into-RobotLocomotion-master_20151008-1111-Continuous

There are 5 that list missing dependency 'SNOPT' and one that lists 'snopt'

To reproduce, simply run the build using WITH_SNOPT off and WITH_SNOPT_PRECOMPILED on.

RussTedrake, 2015-10-08:
@hongkai-dai - could you take a look?

hongkai-dai, 2015-10-08:
Yes, will look into it

hongkai-dai, 2015-10-08:
This is a problem with the old `inverseKinTraj.m`  function. The other IK functions (`InverseKinematics.m`, `inverseKin.cpp`, etc) either has C++ implementation directly (`inverseKin.cpp`, `inverseKinTraj.cpp`), or they are derived from `NonlinearProgram.m`, so they can call `NonlinearProgramsnoptmex`. But `inverseKinTraj.m` still calls `inverseKinBackend.m`, which calls MATLAB snopt directly. This approach does not work with precompiled snopt.

As our goal is to get rid of the `RigidBodyConstraint` class, and uses only `Constraint` class and `NonlinearProgram` as the interface for IK, I suggest not to test the inverseKinTraj.m function when we are using the precompiled version of snopt.

hongkai-dai, 2015-10-08:
The reason why `inverseKinTraj.m`, `inverseKinPointwise.m` and `inverseKin.m` are looking for the MATLAB version of `inverseKinBackend.m`, is that we do not compile `inverseKinmex`, `inverseKinTrajmex` and `inverseKinPointwisemex` with SNOPT.  Is it possible that we can precompile these three files also, as we did for `NonlinearProgramSnoptmex`?

RussTedrake, 2015-10-09:
as discussed, let's just fix the easy ones first.  we eventually want to distribute the drake libs which depend on libsnoptc , but that will take longer I know.

RussTedrake, 2016-05-01:
this will be resolved by #245 and the eventual move to everything running through our c++ solver wrappers.

closed 2016-05-01 by RussTedrake