RobotLocomotion/drake #1415: provide fastqp (especially for solveLCP) even if gurobi is not present

RussTedrake, 2015-10-10:
not everyone has gurobi, and we could pick a more portable fallback.

Lucy-tri, 2016-06-07:
@ggould-tri should I prioritize this lower than the C++ trajectory optimization work (pendulum, DircolTrajectoryOptimization)?

ggould-tri, 2016-06-07:
Yes.  Right now non-emergency issues are a lower priority than current
development.  That may change as we staff up, but not before you are likely
to be done with pendulum anyway.

amcastro-tri, 2016-06-07:
This is great! I just talked to Russ' students this morning just about this. And we even made a little test calling fastQP but of course it failed. It'd be super useful to have that working. Right now Matlab calls fastQP and only if it fails it calls the Gurobi solver.

RussTedrake, 2016-12-23:
mathematical program has this soundly resolved going forward (would fallback to snopt).  still have to finish porting fastQP, but will close this particular issue.

closed 2016-12-23 by RussTedrake