RobotLocomotion/drake #1417: many atlas examples timeout on the build servers

RussTedrake, 2015-10-11:
i've made timeouts = errors on the build servers

http://drake005.csail.mit.edu/drake/pull-RussTedrake-win32-into-RobotLocomotion-master_20151010-2310-Continuous

<matlab_test>strcmp(error_id,"timeout") && ~isempty(strfind(test_name,"Atlas")</matlab_test>

rdeits, 2015-12-14:
On my machine, I have:

runAtlasTurning: 338s
runAtlasTurning270: 286s
runAtlasWalkingTiltedBlocks: 594s
runAtlasWalkingNarrowStairs: 555s

but the build servers are a lot slower. Some of these tests can be truncated, but, for example, the 270 degree turning test is there to check for a bug that can't be exposed with a shorter trajectory. I'll see if I can't speed it up a little...

Of course, we can also just increase the timeouts, but I'd rather not make the builds even slower.

rdeits, 2015-12-14:
About 30% of the entire simulation time is spent in `vToqdot` (in matlab). We have that functionality in C++, but calling it requires some argument changes (vToqdot will need to take a kinsol, not q as its input). I'll look into it. 

solveMexLCP also looks like a function that should be moved to C++. During the turning test, it gets called 36,000 times, but all it actually does is dispatch to a bunch of matlab functions which do their work in C++. It also generates at least one extraneous kinsol by calling doKinematics (with velocities) and then manipulatorDynamics (which calls doKinematics with velocities internally). 

I'd estimate that those two changes would probably double the simulation speed for Atlas.

RussTedrake, 2015-12-27:
will try closing this and see if the timeouts have resolved themselves.

RussTedrake, 2015-12-27:
walking on the tilted blocks still timed out:
http://drake005.csail.mit.edu/drake/pull-hanssusilo-master-into-RobotLocomotion-master_20151227-1809-Continuous

RussTedrake, 2016-05-01:
resolved this by marking them as long running tests (and not doing them during the CI tests)

closed 2016-05-01 by RussTedrake