RobotLocomotion/drake #1418: testIKtraj can timeout on the build servers

RussTedrake, 2015-10-11:
http://drake005.csail.mit.edu/drake/pull-RussTedrake-win32-into-RobotLocomotion-master_20151010-2310-Continuous

<matlab_test>strcmp(error_id,"timeout") && strcmp(test_name,'systems/plants/test/testIKtraj')</matlab_test>

RussTedrake, 2016-01-23:
http://drake004.csail.mit.edu/drake/pull-gizatt-gizatt_signed_dist-into-RobotLocomotion-master_20160122-0338-Continuous

RussTedrake, 2016-01-26:
tests are still failing:
http://drake005.csail.mit.edu/drake/pull-henryhwu-fix-1652-into-RobotLocomotion-master_20160126-1206-Continuous

hongkai-dai, 2016-01-26:
But on CDash, `testIKtraj` passes without timing problem. `testBMI` fails with a segmentation fault. http://kobol.csail.mit.edu/cd/viewTest.php?buildid=20322

PR #1651 (2016-01-24): fix the random seed in testIKtraj and testKinCnst
fix #1418
PR #1659 (2016-01-27): add timeout to testIKtraj
Fix #1418 testIKtraj takes 16 minutes on drake005
closed 2016-01-27 by RussTedrake