RobotLocomotion/drake #1419: RigidBodyManipulator: Calculate gravity forces in joints

steinachim, 2015-10-11:
I need to calculate the torque that is created by gravity in the joints of a `RigidBodyManipulator` (in a fix pose, without robot movement). Is there already something like this implemented in Drake?

`RigidBodyManipulator.drawLCMGLForces()` seems to do something like this, but I do not really understand how to use the RigidBodyForceElements...

Could anybody help? Thank you.

avalenzu, 2015-10-11:
`GravityCompensationTorqueConstraint`  calculates something close to what you want, but you'll need to incorporate the effect of external (contact) forces yourself - they aren't included in that constraint.

tkoolen, 2015-10-12:
The trick that's also exploited in `GravityCompensationTorqueConstraint` is that the gravitational torques are equal to the `C` variable coming out of `manipulatorDynamics` (assuming you want to use Matlab), when you set the velocity vector and external forces to zero.

steinachim, 2015-10-16:
Thank you, this is exactly what I was looking for.

closed 2015-10-16 by steinachim