RobotLocomotion/drake #1422: addpath_spotless points to wrong build folder?

RussTedrake, 2015-10-13:
From Piazza:

After building from source on windows 10 x64, and WITH_SPOTLESS=1, I get:

```
>> acrobotROA
 Calling addpath_spotless
Warning: Name is nonexistent or not a directory:
C:\Users\Eric\Repos\drake-distro\drake\..\build\matlab\spotless 
```

rdeits, 2015-10-13:
So far, i can't reproduce this. I'm running the most up-to-date version of spotless on my machine, and it seems to install the correct addpath_ script. Also, the logic that creates the addpath_spotless file hasn't changed in a while. I'll check in with the original poster on piazza.

raysworld, 2015-12-02:
I have met the same problem and cannot find a solution. I am using Windows 10 x64 build 10586 with Matlab 2015b, and the Make process reported no errors with WITH_SPOTLESS option enabled.
When I tried to run the demo in examples/grasping/forceClosure/test/testSynsizeGrasp.m it reported:

Warning: Name is nonexistent or not a directory:
D:\Software\MATLAB\drake-distro\drake\..\build\matlab\spotless 
> In path (line 109)
  In addpath (line 88)
  In addpath_spotless (line 3)
  In checkDependency>pod_pkg_config (line 453)
  In checkDependency (line 54)
  In testSynthesizeGrasp (line 14) 

Cannot find required dependency: spotless

rdeits, 2015-12-14:
This issue doesn't appear to exist on the build servers, which leads me to believe that it's an issue that only happens outside of cygwin. Do we have any non-cygwin machines to test on?

raysworld, 2015-12-15:
I found there is a persistent variable 'conf' in checkDependency.m. When it is called by examples/grasping/forceClosure/test/testSynsizeGrasp.m it reported the error warning above. However, if I clear this variable and re-run the command checkDependency('spotless'), the check will pass. It seemed that the dependency for 'spotless' had been checked and set as 'not ok'.

Therefore I wonder is there any possibility that this is an initialization issue or something else.

RussTedrake, 2016-05-01:
i believe this was probably the same issue as #1674, and that it has now been resolved in https://github.com/RobotLocomotion/spotless-pod/commit/fccbd2d2bde0ba768bd01da1dbf7c26ed9806994

closed 2016-05-01 by RussTedrake