RobotLocomotion/drake #1423: Migrate pods build system to catkin

edowson, 2015-10-13:
The effort required to develop and maintain the pods-based build scripts can be greatly reduced by switching over to catkin (ROS).

The catkin macros are a super-set of the functionality provided by CMake, and are built on top of CMake.

It's quite straightforward and easy to migrate an existing CMake project, to support builds within a catkin workspace. 

The following tutorial gives a good introduction:
http://jbohren.com/articles/gentle-catkin-intro/

rdeits, 2015-10-13:
That's a very clear tutorial; thanks for sharing it. I'll play around with Catkin on my own, but having *just* overhauled our build system this summer, I don't think we have the momentum or motivation to do it again right now ( @RussTedrake correct me if I'm wrong :wink: ). But catkin looks nice, and I'm interested in learning more about it.

RussTedrake, 2015-12-16:
we believe this is too ubuntu-specific.

closed 2015-12-16 by RussTedrake

edowson, 2015-12-17:
Hi Russ,

That's not quite true. I do most of my development on Mac OS X.

Catkin is just a set of additional CMake macros, and is as portable as CMake.

Regards,

Elvis Dowson