RobotLocomotion/drake #1425: python lcmtypes import error with new lcm-gen

patmarion, 2015-10-14:
Note, this is likely an lcm bug, but I am posting it here first because it can affect drake.  Tested using lcm-gen 1.2.1.  Note, drake currently builds lcm 1.0.

The Python lcmtype generation now imports dependent types using the full package name, instead of just the lcmtype name.  So for example, using lcm-gen 1.2.1 in lcmt_polynomial_matrix.py, it will say:

    import drake.lcmt_polynomial

while lcm-gen 1.0 just had:

    import lcmt_polynomial

The problem is drake lcmtypes are installed with --ppath lcmtypes, so with the full package the import statement should actually be:

    import lcmtypes.drake.lcmt_polynomial

We'll probably need to submit a patch upstream to lcm to fix this.  In the meantime, the drake lcmtypes may be broken in the drc codebase because drc does necessarily use lcm 1.0, it uses the system install version.  I fixed drc by copying lcm-gen from drake's build/bin into drc's build/bin.

RussTedrake, 2015-12-27:
@rdeits -- did this get fixed with your python mods?

rdeits, 2016-01-26:
No, but I gave up on using the `drake` package name in python because of the LCM issues. I think we should revert the change in cmake and stop setting --ppath

patmarion, 2016-02-04:
this can be closed now.  resolved by pull request #1664

closed 2016-02-04 by RussTedrake