RobotLocomotion/drake #1426: Could not find MEX_CC using mex -v

tarquasso, 2015-10-15:
Trying to compile drake, getting stuck at the "configure step for 'drake'"
with this error, any ideas why that might be? Could not find another issue that covered this particular error.

```
-- Up-to-date: /usr/local/lib/pkgconfig/eigen3.pc
[ 74%] No install step for 'eigen'
[ 75%] Completed 'eigen'
[ 78%] Built target eigen
[ 78%] Performing forceconfigure step for 'drake'
Force configure of drake
[ 79%] Performing configure step for 'drake'
-- CMAKE_INSTALL_PREFIX=/Users/rkk/Documents/MATLAB/drake-distro/build
CMake Error at cmake/mex.cmake:36 (message):
  Could not find MEX_CC using mex -v
Call Stack (most recent call first):
  cmake/mex.cmake:162 (get_mex_option)
  CMakeLists.txt:86 (mex_setup)

-- Configuring incomplete, errors occurred!
See also "/Users/rkk/Documents/MATLAB/drake-distro/drake/pod-build/CMakeFiles/CMakeOutput.log".
make[3]: *** [drake-prefix/src/drake-stamp/drake-configure] Error 1
make[2]: *** [CMakeFiles/drake.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2
```

tarquasso, 2015-10-15:
just figured out it has to do with xcode 7 not being supported by matlab. I update xcode 7 today since homebrew was not properly working and brew doctor was telling me to update... Now i need to figure out how to solve this error within matlab:
```
 >> mex -setup
 Error using mex
 No supported compiler or SDK was found. For options, visit
 http://www.mathworks.com/support/compilers/R2015a/maci64.html. 
```

tarquasso, 2015-10-15:
Followed the answer in here: http://www.mathworks.com/matlabcentral/answers/243868-mex-can-t-find-compiler-after-xcode-7-update-r2015b

now was able to make matlab happy again:
```
>> mex -setup
MEX configured to use 'Xcode with Clang' for C language compilation.
Warning: The MATLAB C and Fortran API has changed to support MATLAB
	 variables with more than 2^32-1 elements. In the near future
	 you will be required to update your code to utilize the
	 new API. You can find more information about this at:
	 http://www.mathworks.com/help/matlab/matlab_external/upgrading-mex-files-to-use-64-bit-api.html.

To choose a different language, select one from the following:
 mex -setup C++ 
 mex -setup FORTRAN
>> mex -setup C++
MEX configured to use 'Xcode Clang++' for C++ language compilation.
Warning: The MATLAB C and Fortran API has changed to support MATLAB
	 variables with more than 2^32-1 elements. In the near future
	 you will be required to update your code to utilize the
	 new API. You can find more information about this at:
	 http://www.mathworks.com/help/matlab/matlab_external/upgrading-mex-files-to-use-64-bit-api.html.
```
but problem in drake make still persists, any ideas?

RussTedrake, 2015-10-15:
fwiw - answers to this issue are in the drake FAQ.  can you make sure that the matlab being found by cmake is the version you think?  (e.g., you can look at the pod-build/CMakeCache.txt)

tarquasso, 2015-10-15:
still same error during make.

my CMakeCache.txt shows
```
//Path to a program.
matlab:FILEPATH=/usr/bin/matlab
```

```
$ readlink /usr/bin/matlab
/Applications/MATLAB_R2015a.app/bin/matlab
```

followed the guide on the FAQ,
I adjusted for MATLAB_R2015a.app:

```
edit ([matlabroot '/bin/maci64/mexopts/clang_maci64.xml'])
edit ([matlabroot '/bin/maci64/mexopts/clang++_maci64.xml'])
edit ([matlabroot '/bin/maci64/mexopts/gfortran.xml'])
```

by having now twice the following block in each of those files.
```
                    <dirExists name="$$/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.9.sdk" />
                    <dirExists name="$$/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk" />
                    <dirExists name="$$/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk" />
                    <cmdReturns name="find $$ -name MacOSX10.9.sdk" />
                    <cmdReturns name="find $$ -name MacOSX10.10.sdk" />
                    <cmdReturns name="find $$ -name MacOSX10.11.sdk" />
```

the FAQ also mentioned to edit this:
```
edit ([matlabroot '/bin/maci64/mexopts/intel_fortran.xml'])
```
it was not clear to me what I can change within that file, it looks completely different than the other 3 compiler xml files.

Any suggestions?

tarquasso, 2015-10-15:
Twan helped me, it had to do mostly with a make options bug: 
in make options, the flag:
```
 CMAKE_INSTALL_PREFIX 
```
was pointing to 
```
/usr/local/bin
```
, whereas it should actually point to 
```
/path/to/drake-distro/build  
```
After changing that setting and using the 'c' button to configure, then no 'g' option was showing up to then actually generate and exit the make options menu. we had to hold down the arrow keys while hitting 'c' to make the 'g' button to eventually appear.

With the wrong cmake settings, whenever ``make`` was run in ``drake-distro``, it polluted ``/usr/local/bin`` with subfolders full of header, dylib and .pc package files. This caused homebrew to get a bit messed up. after cleaning the brew doctor errors away, fixing the make options as described above, and then running 

```
./install_prereqs.sh homebrew
```
again, we could eventually run ``make gurobi`` and then ``make -j7`` in drake-distro. The second ``make -j7`` failed a few times, but eventually was able to compile after running it for the third or fourth time...

i am back to normal, thanks Twan.

closed 2015-10-15 by tarquasso