RobotLocomotion/drake #1428: precompiled on mac assumes >= 2014a

RussTedrake, 2015-10-15:
from piazza:

I installed from binary, (v.0.9.11), using Matlab R2012a (7.14.0.739) 64-bit in Snow Leopard 10.6.8.

```
>> addpath_drake
Adding /Users/Tony/Documents/MATLAB/drake_distro/drake/../build/matlab to the matlab path

(cd to examples/CartPole)
>> runLQR
Error using DrakeSystem/getModel (line 231)
Unknown option specified with the generic function of S-function 'DCSFunction' in
'FeedbackSystem_4744833/DrakeSys'. This may be caused by using an S-function generated
in a more recent version of Simulink than the one you are currently using.

Error in DynamicalSystem/simulate (line 38)
mdl = getModel(obj);

Error in DrakeSystem/simulate (line 494)
[varargout{:}] = simulate@DynamicalSystem(obj,varargin{:});

Error in runLQR (line 39)
xtraj=simulate(sys,[0 4],x0+.2*randn(4,1));
```

i think drake001 needs to have 2012a (re-)installed?

psiorx, 2015-10-15:
Didn't we have another reason to move our minimum supported MATLAB version up to R2014a?

RussTedrake, 2015-10-16:
the older versions don't officially support the newer compilers.  but we've been managing just fine (essentially by passing older compiler flag versions to newer compilers), and on edx it's clear that lots of people have older versions of matlab.

RussTedrake, 2016-05-01:
resolving this in anticipation of resolving #1766.

closed 2016-05-01 by RussTedrake