RobotLocomotion/drake #1429: Error in Manipulator dynamics() when running a Dircol solveTraj with quaternion coordinates

crogers3, 2015-10-15:
It seems that when ```Manipulator.dynamics()``` is called from Dircol's ```solveTraj()``` with quaternion coordinates, that rpy is still expected, as there is an inconsistency with dimensions (off by one):
```
Error using horzcat
Dimensions of matrices being concatenated are not consistent.

Error in Manipulator/dynamics (line 62)
          dvdot = [zeros(obj.num_positions,1),...

Error in DircolTrajectoryOptimization/dynamics_data (line 84)
      [data.xdot,data.dxdot] = obj.plant.dynamics(0,x,u);

Error in DircolTrajectoryOptimization>@(varargin)obj.dynamics_data(varargin{:}) (line 39)
        obj = obj.addSharedDataFunction(@obj.dynamics_data,{obj.x_inds(:,i);obj.u_inds(:,i)});
```
The problematic lines are:
```
62:  dvdot = [zeros(obj.num_positions,1),...
63:    -Hinv*matGradMult(dH(:,1:obj.num_positions),vdot) + Hinv*dtau(:,1:obj.num_positions),...
64:    +Hinv*dtau(:,1+obj.num_positions:end), Hinv*B];
```

RussTedrake, 2015-12-27:
@crogers3 -- i believe we fixed this on your machine.  did the fix get pushed up?

crogers3, 2015-12-27:
I don't think we ever fixed this bug on my machine. We had fixed a similar one in bug [1409](https://github.com/RobotLocomotion/drake/issues/1409). None of my fork's branches have a fix for this particular bug.

RussTedrake, 2016-05-01:
@mposa -- could you take a look?  should be a simple enough fix.

jwnimmer-tri, 2017-06-19:
Closing as won't-fix, because #6369 will be removing the relevant software.

If I'm confused and this should in fact stay open, please re-open an clarify the remaining unsolved problem(s) in a new comment here.

closed 2017-06-19 by jwnimmer-tri