RobotLocomotion/drake #1432: drake installation issue

edwardtung, 2015-10-17:
Hello. I am trying to install Drake on mac
these message show up after a make the files 
>>
-- Generating done
-- Build files have been written to: /Users/edwardtung/drake-distro/externals/director/distro/pods/drake-distro/pod-build/src/ddapp-build
[ 81%] Performing build step for 'ddapp'
[  1%] Generating depth_compose.cxx
/bin/sh: ./depth_compose.cxx: No such file or directory
src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/build.make:68: recipe for target 'src/vtk/DRCFilters/depth_compose.cxx' failed
gmake[10]: *** [src/vtk/DRCFilters/depth_compose.cxx] Error 127
CMakeFiles/Makefile2:997: recipe for target 'src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/all' failed
gmake[9]: *** [src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/all] Error 2
Makefile:126: recipe for target 'all' failed
gmake[8]: *** [all] Error 2
CMakeFiles/ddapp.dir/build.make:112: recipe for target 'src/ddapp-stamp/ddapp-build' failed
gmake[7]: *** [src/ddapp-stamp/ddapp-build] Error 2
CMakeFiles/Makefile2:167: recipe for target 'CMakeFiles/ddapp.dir/all' failed
gmake[6]: *** [CMakeFiles/ddapp.dir/all] Error 2
Makefile:75: recipe for target 'all' failed
gmake[5]: *** [all] Error 2
Makefile:26: recipe for target 'all' failed
gmake[4]: *** [all] Error 2
CMakeFiles/director.dir/build.make:113: recipe for target 'director-prefix/src/director-stamp/director-build' failed
gmake[3]: *** [director-prefix/src/director-stamp/director-build] Error 2
CMakeFiles/Makefile2:677: recipe for target 'CMakeFiles/director.dir/all' failed
gmake[2]: *** [CMakeFiles/director.dir/all] Error 2
Makefile:75: recipe for target 'all' failed
gmake[1]: *** [all] Error 2
make: *** [all] Error 2

and addpath_drake under drake wouldn't work 
saids 
>>
Error using addpath_drake (line 24)
You must run make first (and/or add your pod build/matlab directory to the matlab path)

be very grateful to know what's going on, thx.

rdeits, 2015-10-17:
We've seen this before. Check out https://github.com/RobotLocomotion/director/issues/56#issuecomment-132323338

edwardtung, 2015-10-18:
thx mate I'll dig in.

edwardtung, 2015-10-18:
i try to follow#56
encountered the some problem 

$ brew uninstall vtk
Error: No such keg: /usr/local/Cellar/vtk
$ brew install vtk5 --with-qt
==> Installing vtk5 from homebrew/homebrew-science
Error: You must `brew link pyqt' before homebrew/science/vtk5 can be installed
$ brew link pyqt
Linking /usr/local/Cellar/pyqt/4.11.4... Error: Permission denied - /usr/local/lib/python2.7/site-packages/PyQt4

what should i do?

edwardtung, 2015-10-19:
now i manage to run $ brew install vtk5 --with-qt
but when run make in drake-distro got this:

Configuring with CMAKE_FLAGS: -DCMAKE_BUILD_TYPE=Release
-- The C compiler identification is AppleClang 7.0.0.7000072
-- The CXX compiler identification is AppleClang 7.0.0.7000072
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
CMake Error: Generator: execution of make failed. Make command was: "/opt/local/bin/gmake" "cmTC_4a560/fast"
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- broken
CMake Error at /usr/local/Cellar/cmake/3.3.2/share/cmake/Modules/CMakeTestCCompiler.cmake:61 (message):
  The C compiler
  "/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc"
  is not able to compile a simple test program.

 It fails with the following output:

   Change Dir: /Users/edwardtung/drake-distro/pod-build/CMakeFiles/CMakeTmp

  

  Run Build Command:"/opt/local/bin/gmake" "cmTC_4a560/fast"

  No such file or directory

  Generator: execution of make failed.  Make command was:
  "/opt/local/bin/gmake" "cmTC_4a560/fast"
    

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/edwardtung/drake-distro/pod-build/CMakeFiles/CMakeOutput.log".
See also "/Users/edwardtung/drake-distro/pod-build/CMakeFiles/CMakeError.log".
make: *** [configure] Error 1

don't know what's the issue here, till searching...

RussTedrake, 2015-10-19:
that definitely looks like there is something unhealthy about your xcode installation.  cmake can run the compiler successfully on a simple test program.  Perhaps try reinstalling xcode?

edwardtung, 2015-10-19:
oh my...
alright give it a shot

RussTedrake, 2015-12-27:
not sure how much we can help with this one.  let us know if there is still interest and we can help

closed 2015-12-27 by RussTedrake