RobotLocomotion/drake #1440: Error While running make

SainiVinod, 2015-10-24:
Following errors occurred while running Make even after following every steps:

warning: [options] bootstrap class path not set in conjunction with -source 1.6
1 error
1 warning
make[10]: *** [../lcmtypes/java/bot_param/entry_t.class] Error 1
make[9]: *** [CMakeFiles/lcmtypes_bot2-param_jar.dir/all] Error 2
make[8]: *** [all] Error 2
make[7]: *** [bot2-param-prefix/src/bot2-param-stamp/bot2-param-build] Error 2
make[6]: *** [CMakeFiles/bot2-param.dir/all] Error 2
make[5]: *** [all] Error 2
make[4]: *** [all] Error 2
make[3]: *** [libbot-prefix/src/libbot-stamp/libbot-build] Error 2
make[2]: *** [CMakeFiles/libbot.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2

Could you please tell me where I am making any mistake?

RussTedrake, 2015-10-24:
i think you haven’t shown us enough of the output. the actual error probably happened higher up?

SainiVinod, 2015-10-25:
error: error reading /home/vinod/drake-distro/build/share/java/lcmtypes_bot2-param.jar; zip file is empty
warning: [options] bootstrap class path not set in conjunction with -source 1.6
1 error
1 warning
make[10]: *** [../lcmtypes/java/bot_param/entry_t.class] Error 1
make[9]: *** [CMakeFiles/lcmtypes_bot2-param_jar.dir/all] Error 2
make[8]: *** [all] Error 2
make[7]: *** [bot2-param-prefix/src/bot2-param-stamp/bot2-param-build] Error 2
make[6]: *** [CMakeFiles/bot2-param.dir/all] Error 2
make[5]: *** [all] Error 2
make[4]: *** [all] Error 2
make[3]: *** [libbot-prefix/src/libbot-stamp/libbot-build] Error 2
make[2]: *** [CMakeFiles/libbot.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2

RussTedrake, 2015-10-25:
what version of java do you have?
are there any more java errors higher up?

SainiVinod, 2015-10-25:
No, I dont think so. I am using java 1.7.0_79.

SainiVinod, 2015-10-25:
This is the output file after running Make command.

https://drive.google.com/file/d/0B9fvdy2ryC8_cEJhb2toUGhEdjg/view?usp=sharing

Thanks for your help.

RussTedrake, 2015-12-27:
Sorry this fell off my stack.  Let us know if it is still an issue.

The only potentially unusual thing I see here is: 
```
-- gdkx.h not found, system not appear to have gtk+X11 support
-- Configuring without bot2-vis, modules requiring gtk+x11 will not be built.
```
you could disable libbot (in the cmake options, set `WITH_LIBBOT = OFF`), it gives you very little these days.

RussTedrake, 2016-01-23:
please reopen this if you're still stuck.

closed 2016-01-23 by RussTedrake