RobotLocomotion/drake #1441: Error running make

itsamk, 2015-10-24:


RussTedrake, 2015-10-24:
What happens if you manually run the cmake-gui command from the makefile on your command line?

itsamk, 2015-10-24:
I get the c-make GUI. What to set as source of the code and destination of binaries?

RussTedrake, 2015-10-24:
If it is launched correctly, those should be set for you.  Maybe the problem was earlier in the make file for you.  How about running make options but with the debug flags set?

itsamk, 2015-10-24:
[image]

I ran 'make options -d' and this was the output.

RussTedrake, 2015-10-24:
You didn't run it from the correct directory. 

Have you tried the precompiled versions?

itsamk, 2015-10-24:
The precompiled version worked like a charm! :) Thanks!

closed 2015-10-24 by itsamk